riscv-collab/riscv-gnu-toolchain #1016: Failed to build with gcc-5.4.0

Pengwei-Jin, 2022-01-24:
Hi, I want to build riscv-gnu-toolchain with gcc-5.4.0, but failed in build-gcc-linux-stage1. I just git clone this repo and checkout riscv-gcc to the tag "gcc-5_4_0-release". Do I need to do any other steps? Can anyone help me to solve this problem? Part of log is following and the complete log is in the attach file.

Configuring in ./gcc
configure: creating cache ./config.cache
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
checking target system type... riscv32-unknown-linux-gnu
checking LIBRARY_PATH variable... ok
checking GCC_EXEC_PREFIX variable... ok
checking whether to place generated files in the source directory... no
checking whether a default linker was specified... no
checking whether a default assembler was specified... no
checking for x86_64-unknown-linux-gnu-gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for suffix of executables... 
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for x86_64-unknown-linux-gnu-gnatbind... no
checking for gnatbind... no
checking for x86_64-unknown-linux-gnu-gnatmake... no
checking for gnatmake... no
checking whether compiler driver understands Ada... no
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep

......

checking whether getrusage is declared... yes
checking whether ldgetname is declared... no
checking whether times is declared... yes
checking whether sigaltstack is declared... yes
checking whether madvise is declared... yes
checking for struct tms... yes
checking for clock_t... yes
checking for F_SETLKW... yes
checking if mkdir takes one argument... no
Makefile:4046: recipe for target 'configure-gcc' failed
make[1]: Leaving directory '/home/shenwei/riscv32-linux/riscv-gnu-toolchain/build-gcc-linux-stage1'
Makefile:351: recipe for target 'stamps/build-gcc-linux-stage1' failed

[build.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7923013/build.log)

pz9115, 2022-01-24:
As I know this old version 5.4.0 of gcc does not support riscv, you can check availbe gcc version in follow repo-branch:

https://github.com/riscv-collab/riscv-gcc

Pengwei-Jin, 2022-01-24:
The issue #632 has mentioned that gcc-5.4.0 can make riscv-gnu-toolchain without errors. Do I have a misunderstanding?

pz9115, 2022-01-24:
> The issue #632 has mentioned that gcc-5.4.0 can make riscv-gnu-toolchain without errors. Do I have a misunderstanding?

This means you can make riscv-gnu-toolchain use a local compiler on your machine like gcc-5.4.0, but not means to change the source code in riscv-gnu-toolchain/riscv-gcc into gcc-5.4.0.

Pengwei-Jin, 2022-01-24:
> This means you can make riscv-gnu-toolchain use a local compiler on your machine like gcc-5.4.0, but not means to change the source code in riscv-gnu-toolchain/riscv-gcc into gcc-5.4.0.

Oh! I got it. Thank you~

closed 2022-01-24 by Pengwei-Jin