riscv-collab/riscv-gnu-toolchain #1018: Linking error when using -lstdc++

SaraNassef, 2022-01-25:
In my example I use both cpp and c files, compilation using the riscv32-unknown-elf-gcc is done successfully but in the linking step I first got this error related to the object files of some of the cpp files as test.cpp
`.../BUILD/PULP/GCC_RISCV//test.o:(.sdata._ZTI10test[_ZTI10test]+0x0): undefined reference to `vtable for __cxxabiv1::__class_type_info'
`
by adding -lstdc++ to the ld_flags in order to solve the error above, I get this error
`/opt/riscv/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/lib/rv32imcxgap9/ilp32/libstdc++.a(del_op.o): In function `operator delete(void*)':
/home/osboxes/pulp-riscv-gnu-toolchain/riscv-gcc/libstdc++-v3/libsupc++/del_op.cc:48: undefined reference to `free'
collect2: error: ld returned 1 exit status
`
I would appreciate any help with how I can solve this error.

aswaterman, 2022-01-25:
`free` is defined in the C library (i.e. `-lc`).  But I think the issue might be that you're linking with `ld` or `gcc` rather than linking with `g++`.  The latter automatically links against the C++ library in the correct link order.

TommyMurphyTM1234, 2022-01-26:
You probably need to post a more detailed build log showing the exact commands being called to compile and link.

SaraNassef, 2022-01-26:
[build_log.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7941726/build_log.txt)

TommyMurphyTM1234, 2022-01-26:
> [build_log.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7941726/build_log.txt)

That log seems to contain a different error to the ones that you're posting about above?

`/home/osboxes/pulp/pulp-sdk/rtos/pulpos/common/rules/pulpos/default_rules.mk:232: recipe for target '/home/osboxes/VendorUTT/lib/ghost/src/firmware/BUILD/PULP/GCC_RISCV//VendorUTT/VendorUTT' failed`

You also seem to be using `-nostdlib` which doesn't align with your intention to link standard libraries.

SaraNassef, 2022-01-26:
I modified the file, the error line appeared in the terminal but wasn't logged in the file so I added it.

I tried removing the `-nostdlib`  which resulted in a different error `/opt/riscv/lib/gcc/riscv32-unknown-elf/7.1.1/../../../../riscv32-unknown-elf/lib/rv32imcxgap9/ilp32/libgloss.a(syscalls.o):(.sdata+0x0): undefined reference to '_end'
collect2: error: ld returned 1 exit status`

[build_log.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7942142/build_log.txt)

TommyMurphyTM1234, 2022-01-26:
> `undefined reference to '_end'`

Does you linker script provide the `_end` symbol?

SaraNassef, 2022-01-26:
No it doesn't seem to provide it.
[link.ld.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7942382/link.ld.txt)

TommyMurphyTM1234, 2022-01-26:
> No it doesn't seem to provide it. [link.ld.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/7942382/link.ld.txt)

That could be at least part of your problem.
The linker script usually provides certain symbols expected by different standard library mechanisms, such as `sbrk()` in this case.
For example (not RISC-V specific, just for illustrative purposes):

https://users.informatik.haw-hamburg.de/~krabat/FH-Labor/gnupro/7_GNUPro_Embedded_Development/embThe_linker_script.html

SaraNassef, 2022-02-01:
I added _end in my linker script and it worked, with this I can close the issue.

@TommyMurphyTM1234  @aswaterman  Thanks for your help.

closed 2022-02-01 by SaraNassef