riscv-collab/riscv-gnu-toolchain #1019: Faild to compile tool chain

kdpatino, 2022-01-26:
I am trying to compile the toolchain with the following config:

```
./configure --prefix=/gscratch/stf/kpatinos/riscv_tools/riscv --disable-linux --with-arch=rv32i
```
Then when I do the make I got this error:
```
/usr/bin/which: illegal option -- -
../.././riscv-gcc/gcc/config.gcc: line 4690: Usage:: command not found
--with-abi=ilp32d is not supported for ISA 
```

Do you know why I am getting this error?

TommyMurphyTM1234, 2022-01-26:
Please post a full log showing the commands used to clone, configure and make.

kdpatino, 2022-01-26:
Clone Log:

```
git clone https://github.com/riscv/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 7911, done.
remote: Counting objects: 100% (373/373), done.
remote: Compressing objects: 100% (330/330), done.
remote: Total 7911 (delta 54), reused 316 (delta 40), pack-reused 7538
Receiving objects: 100% (7911/7911), 5.48 MiB | 12.07 MiB/s, done.
Resolving deltas: 100% (3774/3774), done.

```

This is the configuration and make log:
```
./configure --prefix=../riscv_tools --disable-linux --with-arch=rv32i
configure: error: expected an absolute directory name for --prefix: ../riscv_tools
kpatinos@n3067:/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain$ pwd
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain
kpatinos@n3067:/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain$ ./configure --prefix=/gscratch/scrubbed/kpatinos/riscv/rv32i --disable-linux --with-arch=rv32i
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/sh
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
kpatinos@n3067:/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain$ make > log
Submodule 'riscv-gcc' (https://github.com/riscv-collab/riscv-gcc.git) registered for path 'riscv-gcc'
Cloning into '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-gcc'...
Submodule 'riscv-newlib' (git://sourceware.org/git/newlib-cygwin.git) registered for path 'riscv-newlib'
Cloning into '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-newlib'...
Submodule 'riscv-binutils' (https://github.com/riscv-collab/riscv-binutils-gdb.git) registered for path 'riscv-binutils'
Cloning into '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils'...
*** This configuration is not supported in the following subdirectories:
     gdbserver readline libdecnumber sim gdb
    (Any other directories should still work fine.)
ar: `u' modifier ignored since `D' is the default (see `U')
configure: WARNING: libdebuginfod is missing or unusable; some features may be unavailable.
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/gas/doc/../../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[5]: [Makefile:963: as.1] Error 255 (ignored)
pod2man: unable to format as.pod
libtool: link: warning: `-version-info/-version-number' is ignored for convenience libraries
libtool: link: warning: `-version-info/-version-number' is ignored for convenience libraries
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/sysinfo.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/defparse.y: warning: 27 shift/reduce conflicts [-Wconflicts-sr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/rcparse.y: warning: 58 shift/reduce conflicts [-Wconflicts-sr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/rcparse.y: warning: 10 reduce/reduce conflicts [-Wconflicts-rr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/mcparse.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:920: addr2line.1] Error 255 (ignored)
pod2man: unable to format addr2line.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
[... 148 lines cut ...]
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/ja.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/pt_BR.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/ru.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/sr.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/sv.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/tr.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/uk.gmo': No such file or directory
[... cut ...]
libtool: install: warning: remember to run `libtool --finish /gscratch/scrubbed/kpatinos/riscv/rv32i/libexec/gcc/riscv32-unknown-elf/11.1.0'
configure: WARNING: decimal float is not supported for this target, ignored
configure: WARNING: fixed-point is not supported for this target, ignored
/usr/bin/which: illegal option -- -
/usr/bin/which: illegal option -- -
/usr/bin/which: illegal option -- -
../.././riscv-gcc/gcc/config.gcc: line 4690: Usage:: command not found
--with-abi=ilp32 is not supported for ISA 
make[1]: *** [Makefile:4368: configure-gcc] Error 1
make: *** [Makefile:521: stamps/build-gcc-newlib-stage1] Error 2
```

kdpatino, 2022-01-27:
I didn't realize I copied more logs than I wanted. If you read the next couples line you will see I fixed the "First Error". Just in case here is the log without the "First Error".
```
kpatinos@n3067:/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain$ ./configure --prefix=/gscratch/scrubbed/kpatinos/riscv/rv32i --disable-linux --with-arch=rv32i
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/sh
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
kpatinos@n3067:/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain$ make > log
Submodule 'riscv-gcc' (https://github.com/riscv-collab/riscv-gcc.git) registered for path 'riscv-gcc'
Cloning into '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-gcc'...
Submodule 'riscv-newlib' (git://sourceware.org/git/newlib-cygwin.git) registered for path 'riscv-newlib'
Cloning into '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-newlib'...
Submodule 'riscv-binutils' (https://github.com/riscv-collab/riscv-binutils-gdb.git) registered for path 'riscv-binutils'
Cloning into '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils'...
*** This configuration is not supported in the following subdirectories:
     gdbserver readline libdecnumber sim gdb
    (Any other directories should still work fine.)
ar: `u' modifier ignored since `D' is the default (see `U')
configure: WARNING: libdebuginfod is missing or unusable; some features may be unavailable.
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/gas/doc/../../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[5]: [Makefile:963: as.1] Error 255 (ignored)
pod2man: unable to format as.pod
libtool: link: warning: `-version-info/-version-number' is ignored for convenience libraries
libtool: link: warning: `-version-info/-version-number' is ignored for convenience libraries
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/sysinfo.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/defparse.y: warning: 27 shift/reduce conflicts [-Wconflicts-sr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/rcparse.y: warning: 58 shift/reduce conflicts [-Wconflicts-sr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/rcparse.y: warning: 10 reduce/reduce conflicts [-Wconflicts-rr]
/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/mcparse.y: warning: 1 shift/reduce conflict [-Wconflicts-sr]
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:920: addr2line.1] Error 255 (ignored)
pod2man: unable to format addr2line.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:927: ar.1] Error 255 (ignored)
pod2man: unable to format ar.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:934: dlltool.1] Error 255 (ignored)
pod2man: unable to format dlltool.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:941: nm.1] Error 255 (ignored)
pod2man: unable to format nm.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:948: objcopy.1] Error 255 (ignored)
pod2man: unable to format objcopy.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:955: objdump.1] Error 255 (ignored)
pod2man: unable to format objdump.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
make[4]: [Makefile:962: ranlib.1] Error 255 (ignored)
pod2man: unable to format ranlib.pod
usage: /gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/binutils/../etc/texi2pod.pl [-D toggle...] [infile [outfile]]
[... 130 lines cut ...]
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/ja.gmo': No such file or directory
/usr/bin/install: cannot stat '/gscratch/scrubbed/kpatinos/riscv/riscv-gnu-toolchain/riscv-binutils/ld/po/pt_BR.gmo': No such file or directory
[... cut ...]
libtool: install: warning: remember to run `libtool --finish /gscratch/scrubbed/kpatinos/riscv/rv32i/libexec/gcc/riscv32-unknown-elf/11.1.0'
configure: WARNING: decimal float is not supported for this target, ignored
configure: WARNING: fixed-point is not supported for this target, ignored
/usr/bin/which: illegal option -- -
/usr/bin/which: illegal option -- -
/usr/bin/which: illegal option -- -
../.././riscv-gcc/gcc/config.gcc: line 4690: Usage:: command not found
--with-abi=ilp32 is not supported for ISA 
make[1]: *** [Makefile:4368: configure-gcc] Error 1
make: *** [Makefile:521: stamps/build-gcc-newlib-stage1] Error 2
```

closed 2022-01-27 by kdpatino

TommyMurphyTM1234, 2022-01-27:
What was the root cause?
It may help others who hit the same problem.

kdpatino, 2022-01-27:
This is related to my server environment, I am using singularity containers, don't know why I have issues with `which` program, check this was the hint: 
```
/usr/bin/which: illegal option -- -
```
I don't know why `which` fails using singularity definitions of ubuntu, so I copy a binary from my machine and replace it. So now `which` is working normally and I was able to run the compiler without any issue.