riscv-collab/riscv-gnu-toolchain #1021: Using `riscv64-unknown-elf-g++` to link rust static lib failed

zzhengzhuo, 2022-02-07:
I try to use riscv64-unknown-elf-g++ which is rv64imac architecture to compile C code with rust static lib. 
The simple example is https://github.com/zzhengzhuo/base_bin . 
When I use riscv64-unknown-elf-gcc, it works well. But when I use riscv64-unknown-elf-g++, it failed.

The compile commands like this:
```
make rust
make build/base_bin  
```
The errors like this:
```
riscv64-unknown-elf-g++ -Ibase_lib -v -o build/base_bin main.c base_lib/target/riscv64imac-unknown-none-elf/release/libbase_lib.a 
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-g++
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/zhengzhuo/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --enable-multilib --with-abi=lp64 --with-arch=rv64imac --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 11.1.0 (GCC) 
COLLECT_GCC_OPTIONS='-I' 'base_lib' '-v' '-o' 'build/base_bin' '-mtune=rocket' '-march=rv64imac' '-mabi=lp64' '-march=rv64imac' '-dumpdir' 'build/base_bin-'
 /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/cc1plus -quiet -v -I base_lib main.c -quiet -dumpdir build/base_bin- -dumpbase main.c -dumpbase-ext .c -mtune=rocket -march=rv64imac -mabi=lp64 -march=rv64imac -version -o /tmp/ccpwBEao.s
GNU C++17 (GCC) version 11.1.0 (riscv64-unknown-elf)
        compiled by GNU C version 9.4.0, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/opt/riscv/riscv64-unknown-elf/usr/local/include"
ignoring duplicate directory "/opt/riscv/riscv64-unknown-elf/include"
#include "..." search starts here:
#include <...> search starts here:
 base_lib
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/riscv64-unknown-elf
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include/c++/11.1.0/backward
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/include
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/include-fixed
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C++17 (GCC) version 11.1.0 (riscv64-unknown-elf)
        compiled by GNU C version 9.4.0, GMP version 6.2.0, MPFR version 4.0.2, MPC version 1.1.0, isl version none
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: 70c8dfad8a0cf9a76df8ead2812f5775
COLLECT_GCC_OPTIONS='-I' 'base_lib' '-v' '-o' 'build/base_bin' '-mtune=rocket' '-march=rv64imac' '-mabi=lp64' '-march=rv64imac' '-dumpdir' 'build/base_bin-'
 /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/as -v -I base_lib --traditional-format -march=rv64imac -march=rv64imac -mabi=lp64 -o /tmp/cc4pRmao.o /tmp/ccpwBEao.s
GNU assembler version 2.37 (riscv64-unknown-elf) using BFD version (GNU Binutils) 2.37
COMPILER_PATH=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/libexec/gcc/riscv64-unknown-elf/:/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/lib/gcc/riscv64-unknown-elf/:/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/
LIBRARY_PATH=/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/:/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/:/opt/riscv/riscv64-unknown-elf/lib/
COLLECT_GCC_OPTIONS='-I' 'base_lib' '-v' '-o' 'build/base_bin' '-mtune=rocket' '-march=rv64imac' '-mabi=lp64' '-march=rv64imac' '-dumpdir' 'build/base_bin.'
 /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/collect2 -plugin /opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/liblto_plugin.so -plugin-opt=/opt/riscv/libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccQyuKYm.res -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgloss -plugin-opt=-pass-through=-lgcc --sysroot=/opt/riscv/riscv64-unknown-elf -melf64lriscv -o build/base_bin /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib/crt0.o /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/crtbegin.o -L/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0 -L/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/lib -L/opt/riscv/riscv64-unknown-elf/lib /tmp/cc4pRmao.o base_lib/target/riscv64imac-unknown-none-elf/release/libbase_lib.a -lstdc++ -lm -lgcc --start-group -lc -lgloss --end-group -lgcc /opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/crtend.o
/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/cc4pRmao.o: in function `.L0 ':
main.c:(.text+0xa): undefined reference to `echo(unsigned int)'
collect2: error: ld returned 1 exit status
make: *** [Makefile:5: build/base_bin] Error 1
```

I don't known whether I'm supposed to make the issue here.
But thanks for any help.

TommyMurphyTM1234, 2022-02-07:
> ```main.c:(.text+0xa): undefined reference to `echo(unsigned int)'
> collect2: error: ld returned 1 exit status
> make: *** [Makefile:5: build/base_bin] Error 1```
 
Seems to be a straightforward undefined reference link error.
Why do you think that this is a tools issue rather than simply a problem with the code being compiled/linked?

zzhengzhuo, 2022-02-07:
> Seems to be a straightforward undefined reference link error. Why do you think that this is a tools issue rather than simply a problem with the code being compiled/linked?

Because the code is the simplest code and using `riscv-unkown-elf-gcc` works well.
In my opinion, g++ will be compatible for gcc, so I think there may be something wrong for `riscv-unkown-elf-g++`.

kito-cheng, 2022-02-07:
I guess what you need is `extern "C"` in your header file, more details see https://stackoverflow.com/questions/1041866/what-is-the-effect-of-extern-c-in-c

TommyMurphyTM1234, 2022-02-07:
> In my opinion, g++ will be compatible with gcc

What does that mean?
Anyway, I suspect that @kito-cheng has identified the problem in your code that is causing the link error.

zzhengzhuo, 2022-02-07:
I remember that I tried to add `extern "C"` before.
But when I try it again now, after `cargo clean`, it works -_-!.
I don't known which step was wrong.
Thanks for your time and attention

closed 2022-02-07 by zzhengzhuo