riscv-collab/riscv-gnu-toolchain #1022: -mabi=ilp32f is not supporting error

Sharanabasavaexaleap, 2022-02-09:
Hello,
i need to build riscv32- linux-toolcahin with mabi=ilp32f, but i am getting error while in build. i mentioned steps what i fallowed while build.
step1: from Github : https://github.com/riscv-collab/riscv-gnu-toolchain
step2: i copied code cloned to empty directory as: 
git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
Step 3: i created a folder for build and prefix to that folder and i need for 32 bit used flags as:
 ./configure --prefix=/home/toolchain_32bit/bin --with-arch=rv32imafdc --with-abi=ilp32f
Step4: i need for 32bit linux  so i used: make linux
.....................................................................................................................................................................................................................
**i am getting error as:**
glibc does not yet support the single floating-point ABI
mips nios2 powerpc riscv Makefile:302: recipe for target 'stamps/build-glibc-linux-headers' failed
make: *** [stamps/build-glibc-linux-headers] Error 1
make: *** Waiting for unfinished jobs....
glibc does not yet support the single floating-point ABI
Makefile:319: recipe for target 'stamps/build-glibc-linux-rv32imafdc-ilp32f' failed
make: *** [stamps/build-glibc-linux-rv32imafdc-ilp32f] Error 1

.....................................................................................................................................................................................................................
here in Github https://github.com/riscv-collab/riscv-gnu-toolchain they mentioned as:
[(https://github.com/riscv-collab/riscv-gnu-toolchain#installation-newliblinux-multilib)]
Supported architectures are rv32i or rv64i plus standard extensions (a)tomics, (m)ultiplication and division, (f)loat, (d)ouble, or (g)eneral for MAFD.

Supported ABIs are ilp32 (32-bit soft-float), ilp32d (32-bit hard-float), ilp32f (32-bit with single-precision in registers and double in memory, niche use only), lp64 lp64f lp64d (same but with 64-bit long and pointers).

[image]

TommyMurphyTM1234, 2022-02-10:
What happens if you build from the upstream projects (GCC, gdb, binutils, glibc, etc.) rather than this repo? Maybe using crosstool-NG?

https://crosstool-ng.github.io/

jim-wilson, 2022-03-07:
The linux kernel and glibc only support double-float or no-float.  They both lack support for single-float targets, i.e. no support for F but not D, and no support for ilp32f or lp64f.  So there is no linux support for single-float targets at this time.  You can build embedded elf single-float.  If you do need linux single-float support, then you can try writing the missing glibc and linux kernel support.  That is likely a non-trivial amount of work.  There may also be other stuff missing, like qemu support and riscv-pk support.

Sharanabasavaexaleap, 2022-03-08:
> 

Thank you! Jim-wilson  and tommuMurplhy for your input.

cmuellner, 2022-09-29:
I consider this as "won't fix" as it is a restriction, that is implied by upstream projects.
If someone cares about that, then patches have to be contributed to upstream projects.

closed 2022-09-29 by cmuellner

Sharanabasavaexaleap, 2022-09-29:
> @cmuellner okay