riscv-collab/riscv-gnu-toolchain #1024: Assignment from C variable to inline local register is not working

zetalog, 2022-02-10:
Hi, guys

GCC supports local register variables:
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/new
{code}
register int *p1 asm ("r0") = …;
{code}
However we tested it in forced inline functions:
{code}
#define INSN_DCACHE_FLUSH_ADDR(reg)     (0x9C30002B | ((reg) << 15))

static __inline void vaisra_flush_dcache_addr(void *addr)
{
        __unused register volatile uint64_t a0 asm ("x10") = (uint64_t)addr;

        asm volatile (
                "       add     %0, %0, 0\n"
                "       .word   %1"
                : "=r" (a0) : "i" (INSN_DCACHE_FLUSH_ADDR(10)) : "memory");
}
{code}

And invoke the function from the following piece of test code:
{code}
static void test(void)
{
        uint32_t *addr = (uint32_t *)0x12345600;
        int i;

        for (i = 0; i < 10; i++) {
                *(addr++) = i;
        }
        vaisra_flush_dcache_addr(addr);
}
{code}
And obtained the following assembly:
{code}
000000080000e8c4 <test>:
                *(addr++) = i;
   80000e8c4:   123457b7                lui     a5,0x12345
   80000e8c8:   00100713                li      a4,1
   80000e8cc:   60e7a223                sw      a4,1540(a5) # 12345604 <__text_loc-0x7edcba9fc>
   80000e8d0:   00200713                li      a4,2
   80000e8d4:   60e7a423                sw      a4,1544(a5)
   80000e8d8:   00300713                li      a4,3
   80000e8dc:   60e7a623                sw      a4,1548(a5)
   80000e8e0:   00400713                li      a4,4
   80000e8e4:   60e7a823                sw      a4,1552(a5)
   80000e8e8:   00500713                li      a4,5
   80000e8ec:   60e7aa23                sw      a4,1556(a5)
   80000e8f0:   00600713                li      a4,6
   80000e8f4:   60e7ac23                sw      a4,1560(a5)
   80000e8f8:   00700713                li      a4,7
   80000e8fc:   60e7ae23                sw      a4,1564(a5)
   80000e900:   00800713                li      a4,8
   80000e904:   62e7a023                sw      a4,1568(a5)
   80000e908:   00900713                li      a4,9
   80000e90c:   6007a023                sw      zero,1536(a5)
   80000e910:   62e7a223                sw      a4,1572(a5)
   80000e914:   00050513                mv      a0,a0 <- issue
   80000e918:   9c35002b                0x9c35002b
}
   80000e91c:   00008067                ret
{code}

The compiler generates the following line into:
{code}
__unused register volatile uint64_t a0 asm ("x10") = (uint64_t)addr;
   80000e914:   00050513                mv      a0,a0 <- issue
{code}
However, in the inline context, addr wasn't in a0, but in partially in a4, and actually, addr doesn't exist in any register.

We tested the code using the following riscv-gnu-toolchains:
Linux kernel standard testing toolchain:
riscv64-linux-gcc (GCC) 8.1.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

riscv github repo version:
riscv64-unknown-linux-gnu-gcc (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

aswaterman, 2022-02-10:
From a glance, it looks like there's a bug in your inline asm.  You're using a0 as an output operand (that's what the `=r` means), rather than as an input operand.  You might get what you want if you replace

    : "=r" (a0) : "i" (INSN_DCACHE_FLUSH_ADDR(10)) : "memory");

with
    : : "r" (a0), "i" (INSN_DCACHE_FLUSH_ADDR(10)) : "memory");

zetalog, 2022-02-10:
Oops!
Thanks for the help.
I'll try it again.

closed 2022-02-10 by zetalog