riscv-collab/riscv-gnu-toolchain #1025: how to compile when riscv-gcc is at branch riscv-gcc-10.2.0-rvb

erdevonmatteo11, 2022-02-10:
Hi!
I always meet compile errors when I switch **riscv-gcc to branch riscv-gcc-10.2.0-rvb**. Meanwhile I switch riscv-binutils to the latest branch 2.38 or others like 2.37 and 2.35-rvb, but the compiler still report errors.

TommyMurphyTM1234, 2022-02-10:
What errors?
Build log?

TommyMurphyTM1234, 2022-02-11:
You really should attach the logs rather than embedding them.
And not use screenshot images.
All I can see is config output.
I don't see any build log for the tools.

jim-wilson, 2022-03-07:
The rvb branch is obsolete and no longer maintained.  It is probably implementing a draft version of the B spec incompatible with the ratified spec.  You should use upstream FSF Binutils and GCC sources instead.  None of the branches in riscv-binutils or riscv-gcc are useful other than the default ones checked out by riscv -gnu-toolchain.

The screenshot doesn't show the error.  You probably did a parallel make the the real error scrolled off the screen.  You can cd into the build-gcc-linux-stage1 dir and run make to see the error again.

The config.log output doesn't appear to be useful.

TommyMurphyTM1234, 2022-09-27:
Obsolete branch used and insufficient info provided to show any RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234