riscv-collab/riscv-gnu-toolchain #1028: Problem creating rv32gv

Dm1998, 2022-02-22:
I am trying to create riscv32 gnu toolchain with vector extension following the instructions but everytime I reach that point that it prints those messages " Assembler messages: Error: cannot find default versions of the ISA extension `v' ".
I am followed the instructions from readme and so I made the installation directory, added on $PATH.
$ ../configure --prefix=/home/mylo/gcc22/ --with-arch=rv32gv

```
FLAGS_FOR_BUILD=-g -O2" "CFLAGS_FOR_TARGET=-O2 -D_POSIX_MODE   -mcmodel=medlow" "INSTALL=/usr/bin/install -c" "LDFLAGS=" "LIBCFLAGS=-O2 -D_POSIX_MODE   -mcmodel=medlow" "LIBCFLAGS_FOR_TARGET=-O2 -D_POSIX_MODE   -mcmodel=medlow" "MAKE=make" "MAKEINFO=makeinfo --split-size=5000000 --split-size=5000000   " "PICFLAG=" "PICFLAG_FOR_TARGET=" "SHELL=/bin/bash" "EXPECT=expect" "RUNTEST=runtest" "RUNTESTFLAGS=" "exec_prefix=/home/mylo/gcc22" "infodir=/home/mylo/gcc22/share/info" "libdir=/home/mylo/gcc22/lib" "prefix=/home/mylo/gcc22" "tooldir=/home/mylo/gcc22/riscv32-unknown-elf" "top_toollibdir=/home/mylo/gcc22/riscv32-unknown-elf/lib" "AR=riscv32-unknown-elf-ar" "AS=riscv32-unknown-elf-as" "CC=riscv32-unknown-elf-gcc -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/ -isystem /home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/mylo/rvv.0.8.x/riscv-newlib/newlib/libc/include -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/mylo/rvv.0.8.x/riscv-newlib/libgloss/riscv32   " "LD=riscv32-unknown-elf-ld" "LIBCFLAGS=-O2 -D_POSIX_MODE   -mcmodel=medlow" "NM=riscv32-unknown-elf-nm" "PICFLAG=" "RANLIB=riscv32-unknown-elf-ranlib" "DESTDIR=" \
                        CFLAGS="-O2 -D_POSIX_MODE   -mcmodel=medlow ${flags}" \
                        CCASFLAGS="-O2 -D_POSIX_MODE   -mcmodel=medlow ${flags}" \
                        FCFLAGS=" ${flags}" \
                        FFLAGS=" ${flags}" \
                        ADAFLAGS=" ${flags}" \
                        prefix="/home/mylo/gcc22" \
                        exec_prefix="/home/mylo/gcc22" \
                        GCJFLAGS=" ${flags}" \
                        GOCFLAGS="-O2 -g ${flags}" \
                        CXXFLAGS="-O2 -D_POSIX_MODE   -mcmodel=medlow ${flags}" \
                        LIBCFLAGS="-O2 -D_POSIX_MODE   -mcmodel=medlow ${flags}" \
                        LIBCXXFLAGS="-O2 -D_POSIX_MODE   -mcmodel=medlow -fno-implicit-templates ${flags}" \
                        LDFLAGS=" ${flags}" \
                        MULTIFLAGS="${flags}" \
                        DESTDIR="" \
                        INSTALL="/usr/bin/install -c" \
                        INSTALL_DATA="/usr/bin/install -c -m 644" \
                        INSTALL_PROGRAM="/usr/bin/install -c" \
                        INSTALL_SCRIPT="/usr/bin/install -c" \
                        all); then \
          true; \
        else \
          exit 1; \
        fi; \
      else true; \
      fi; \
    fi; \
  done; \
fi
make[5]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib'
Making all in libc
make[5]: Entering directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc'
Making all in argz
make[6]: Entering directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/argz'
riscv32-unknown-elf-gcc -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/ -isystem /home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/mylo/rvv.0.8.x/riscv-newlib/newlib/libc/include -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/mylo/rvv.0.8.x/riscv-newlib/libgloss/riscv32    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"4.1.0\" -DPACKAGE_STRING=\"newlib\ 4.1.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/argz -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-dummy.o `test -f 'dummy.c' || echo '/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/argz/'`dummy.c
Assembler messages:
Error: cannot find default versions of the ISA extension `v'
/tmp/ccyEZdNm.s:3: Error: cannot find default versions of the ISA extension `v'
make[6]: *** [Makefile:393: lib_a-dummy.o] Error 1
make[6]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/argz'
Making all in stdlib
make[6]: Entering directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/stdlib'
riscv32-unknown-elf-gcc -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/ -isystem /home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/mylo/rvv.0.8.x/riscv-newlib/newlib/libc/include -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/mylo/rvv.0.8.x/riscv-newlib/libgloss/riscv32    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"4.1.0\" -DPACKAGE_STRING=\"newlib\ 4.1.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/stdlib -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-__adjust.o `test -f '__adjust.c' || echo '/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/stdlib/'`__adjust.c
Assembler messages:
Error: cannot find default versions of the ISA extension `v'
/tmp/ccEfIQGO.s:3: Error: cannot find default versions of the ISA extension `v'
make[6]: *** [Makefile:632: lib_a-__adjust.o] Error 1
make[6]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/stdlib'
Making all in ctype
make[6]: Entering directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/ctype'
riscv32-unknown-elf-gcc -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/ -isystem /home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/mylo/rvv.0.8.x/riscv-newlib/newlib/libc/include -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/mylo/rvv.0.8.x/riscv-newlib/libgloss/riscv32    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"4.1.0\" -DPACKAGE_STRING=\"newlib\ 4.1.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/ctype -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-ctype_.o `test -f 'ctype_.c' || echo '/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/ctype/'`ctype_.c
Assembler messages:
Error: cannot find default versions of the ISA extension `v'
/tmp/ccCGgjZe.s:3: Error: cannot find default versions of the ISA extension `v'
make[6]: *** [Makefile:537: lib_a-ctype_.o] Error 1
make[6]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/ctype'
Making all in search
make[6]: Entering directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/libc/search'
riscv32-unknown-elf-gcc -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/ -isystem /home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib/targ-include -isystem /home/mylo/rvv.0.8.x/riscv-newlib/newlib/libc/include -B/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/riscv32 -L/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/libgloss/libnosys -L/home/mylo/rvv.0.8.x/riscv-newlib/libgloss/riscv32    -DPACKAGE_NAME=\"newlib\" -DPACKAGE_TARNAME=\"newlib\" -DPACKAGE_VERSION=\"4.1.0\" -DPACKAGE_STRING=\"newlib\ 4.1.0\" -DPACKAGE_BUGREPORT=\"\" -DPACKAGE_URL=\"\" -I. -I/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/search -fno-builtin -DHAVE_NANOSLEEP      -O2 -D_POSIX_MODE   -mcmodel=medlow -c -o lib_a-bsearch.o `test -f 'bsearch.c' || echo '/home/mylo/rvv.0.8.x/build/../riscv-newlib/newlib/libc/search/'`bsearch.c
Assembler messages:
Error: cannot find default versions of the ISA extension `v'
/tmp/cc0jmpLI.s:3: Error: cannot find default versions of the ISA extension `v'
[... cut ...]
make[4]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib'
make[3]: *** [Makefile:452: all] Error 2
make[3]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib/riscv32-unknown-elf/newlib'
make[2]: *** [Makefile:8492: all-target-newlib] Error 2
make[2]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/mylo/rvv.0.8.x/build/build-newlib'
make: *** [Makefile:567: stamps/build-newlib] Error 2
```

TommyMurphyTM1234, 2022-02-24:
I thought that the recommendation was to use llvm rather than gcc for vector extension support because gcc support is outdated, not maintained and may be broken?

zhongjuzhe, 2022-03-29:
If you want to latest RVV1.0 intrinsic+auto-vectorization support. You need to checkout `riscv-gcc-rvv-next` in `riscv-gcc` submodule and 'riscv-binutils-2.38' in 'riscv-binutils-gdb' submodule. Then build up the whole toolchain.

TommyMurphyTM1234, 2022-09-29:
> If you want to latest RVV1.0 intrinsic+auto-vectorization support. You need to checkout branch `riscv-gcc-rvv-next` in `riscv-gcc` submodule and branch `riscv-binutils-2.38` in `riscv-binutils-gdb` submodule. Then build up the whole toolchain.

I think that this advice has become stale/incorrect as there is no longer any `riscv-gcc-rvv-next` branch. There **is** an `rvv-next` branch where development is ongoing to add/enhance support for the ratified v1.0 V extension.

* https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next

Is the correct way to build the latest V enabled GNU toolchain now to just switch to that branch at the top level and then build as normal?

`git clone https://github.com/riscv/riscv-gnu-toolchain --branch rvv-next`

Or are other steps required? Maybe @zhongjuzhe or @kito-cheng might be able to advise? Thanks.

TommyMurphyTM1234, 2022-09-30:
I am using Zorin OS Lite 16.1 (admittedly not an "officially" supported build host but it is a derivative of Ubuntu 20.04 LTS so I would expect it to work and it has done to date).

I did the following and a build of the newlib bare metal toolchain for `rv64gv` worked fine:
````
cd ~/Downloads
rm -rf riscv-gnu-toolchain
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-toolchain --with-arch=rv64gv
make
````
But when I try to target `rv32gv` it fails:
````
cd ~/Downloads
rm -rf riscv-gnu-toolchain
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-toolchain --with-arch=rv32gv
make
````
I'm attaching the log but I cannot see any error other than this at the end of the log:
```
echo timestamp > s-emit
g++  -fno-PIE -c   -g -O2   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -I. -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/. -I../.././riscv-gcc/gcc/../include -I../.././riscv-gcc/gcc/../libcpp/include -I../.././riscv-gcc/gcc/../libcody  -I../.././riscv-gcc/gcc/../libdecnumber -I../.././riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././riscv-gcc/gcc/../libbacktrace   -o insn-emit.o -MT insn-emit.o -MMD -MP -MF ./.deps/insn-emit.TPo insn-emit.cc
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [Makefile:1143: insn-emit.o] Error 1
make[2]: Leaving directory '/home/user/Downloads/riscv-collab/riscv-gnu-toolchain-rvv32/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4607: all-gcc] Error 2
make[1]: Leaving directory '/home/user/Downloads/riscv-collab/riscv-gnu-toolchain-rvv32/build-gcc-newlib-stage1'
make: *** [Makefile:546: stamps/build-gcc-newlib-stage1] Error 2
```
[build.log.zip](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9687457/build.log.zip)

TommyMurphyTM1234, 2022-09-30:
> I think that this advice has become stale/incorrect as there is no longer any `riscv-gcc-rvv-next` branch. There **is** an `rvv-next` branch where development is ongoing to add/enhance support for [...]

I tested this - building for rv64gv` using the `rvv-next` branch - and it works fine.
```
cd ~/Downloads
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-next
cd riscv-gnu-toolchain
./configure --with-arch=rv64gv --prefix=`pwd`/installed-toolchain
make
```
When I try to target `rv32gv` it fails but I suspect that that's because the machine that I'm using runs out of memory (it only has 4GB!).
I'll try to target `rv32gv` on a better spec'd machine when I get a chance.
```

TommyMurphyTM1234, 2022-10-02:
I tried the `rv32gv` build from the `rvv-next` branch on a machine with 16GB of RAM and it also completed fine.

closed 2022-10-02 by TommyMurphyTM1234