riscv-collab/riscv-gnu-toolchain #1031: Unexpected failure in gcc.dg/Wstringop-overflow-43.c for rv32 targets

palmer-dabbelt, 2022-02-23:
I haven't looked any more closely, but after cleaning up the QEMU submodule I'm seeing an unexpected test failure on rv32.  Not quite sure when I'll have time to look at it, so I figured I'd open the bug to see if anyone else was looking.

````
$ make report
/scratch/palmer/riscv-gnu-toolchain/build/../scripts/testsuite-filter gcc glibc /scratch/palmer/riscv-gnu-toolchain/build/../test/allowlist `find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`
                === gcc: Unexpected fails for rv32imac ilp32 medlow ===
XPASS: gcc.dg/Wstringop-overflow-43.c pr????? (test for warnings, line 172)
                === gcc: Unexpected fails for rv32imafdc ilp32d medlow ===
XPASS: gcc.dg/Wstringop-overflow-43.c pr????? (test for warnings, line 172)

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |
 rv32imafdc/ ilp32d/ medlow |    1 /     1 |    0 /     0 |    0 /     0 |
   rv32imac/  ilp32/ medlow |    1 /     1 |    0 /     0 |    0 /     0 |
   rv64imac/   lp64/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |
make: *** [Makefile:935: report-gcc-linux] Error 1
````

TommyMurphyTM1234, 2022-11-15:
@palmer-dabbelt  - can you clarify the steps needed to attempt to reproduce this?
I have tried the following but I don't know if it's correct - e.g. do I have to build a multilib toolchain, pass additional options to the test suite, etc?
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1031
cd issue-1031
./configure --prefix=`pwd`/installed-tools --with-sim=qemu
make linux
make report-linux SIM=qemu
...
mkdir -p stamps/
date > stamps/check-gcc-linux
/home/tommy/Downloads/riscv-tools/issue-1031/scripts/testsuite-filter gcc glibc /home/tommy/Downloads/riscv-tools/issue-1031/test/allowlist `find build-gcc-linux-stage2/gcc/testsuite/ -name *.sum |paste -sd "," -`

               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
 rv64imafdc/  lp64d/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |
```
Also - I've noticed some possibly strange things in the test logs - perhaps these need to be logged separately?
```
gcc.log
-------

spawn -ignore SIGHUP /home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/gcc/xgcc -B/home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/gcc/ /home/tommy/Downloads/riscv-tools/issue-1031/gcc/gcc/testsuite/gcc.target/riscv/arch-1.c -march=rv64imafdc -mabi=lp64d -mcmodel=medlow -fdiagnostics-plain-output -O0 -march=rv32i -march=rv32I -mabi=ilp32 -S -o arch-1.s
xgcc: error: '-march=rv32I': first ISA subset must be 'e', 'i' or 'g'

spawn -ignore SIGHUP /home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/gcc/xgcc -B/home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/gcc/ offload_gcn663752.c -march=rv64imafdc -mabi=lp64d -mcmodel=medlow -fdiagnostics-plain-output -foffload=amdgcn-amdhsa -S -o offload_gcn663752.s
xgcc: error: GCC is not configured to support 'amdgcn-amdhsa' as '-foffload=' argument
xgcc: note: valid '-foffload=' arguments are: default disable

g++.log
-------

spawn -ignore SIGHUP /home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/gcc/testsuite/g++/../../xg++ -B/home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/gcc/testsuite/g++/../../ offload_gcn1312198.c -march=rv64imafdc -mabi=lp64d -mcmodel=medlow -fdiagnostics-plain-output -nostdinc++ -I/home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/include/riscv64-unknown-linux-gnu -I/home/tommy/Downloads/riscv-tools/issue-1031/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libstdc++-v3/include -I/home/tommy/Downloads/riscv-tools/issue-1031/gcc/libstdc++-v3/libsupc++ -I/home/tommy/Downloads/riscv-tools/issue-1031/gcc/libstdc++-v3/include/backward -I/home/tommy/Downloads/riscv-tools/issue-1031/gcc/libstdc++-v3/testsuite/util -fmessage-length=0 -foffload=amdgcn-amdhsa -S -o offload_gcn1312198.s^M
xg++: error: GCC is not configured to support 'amdgcn-amdhsa' as '-foffload=' argument^M
xg++: note: valid '-foffload=' arguments are: default disable^M
compiler exited with status 1
```

cmuellner, 2022-11-15:
@TommyMurphyTM1234 you need to build/test rv32 (`--with-arch=rv32gc --with-abi=ilp32d`) not rv64.

However, this ticket is from Feb 23 and we bumped all components since then.
Also in the GCC bump in August (see #1128) the allowlist was refreshed.

cmuellner, 2022-11-15:
A clean build of `master` with `--with-arch=rv32gc --with-abi=ilp32d` gives the following test results:

```
               ========= Summary of gcc testsuite =========
                            | # of unexpected case / # of unique unexpected case
                            |          gcc |          g++ |     gfortran |
     rv32gc/ ilp32d/ medlow |    0 /     0 |    0 /     0 |    0 /     0 |

```

Whatever that was back then, it is gone now.

Thanks for reporting, Palmer!

cmuellner, 2022-11-15:
@TommyMurphyTM1234: The errors seem to come from expected fails in the test suite. E.g. in GCC the test file `gcc/testsuite/gcc.target/riscv/arch-1.c` tests for an error in case `-march=rv32I` is provided.

closed 2022-11-15 by cmuellner

TommyMurphyTM1234, 2022-11-15:
Thanks for the update and for closing it @cmuellner. I was still in the middle of building the rv32gc toolchain and you got there before me! 😄