riscv-collab/riscv-gnu-toolchain #1034: Error when running riscv64-unknown-elf-gdb:  riscv64-unknown-elf-gdb: error while loading shared libraries: libpython3.8.so.1.0: cannot open shared object file: No such file or directory

KaminariOS, 2022-03-02:
DISTRIB_ID=ManjaroLinux
DISTRIB_RELEASE=21.2.4

Workaround: 
```bash
cp   /usr/lib/libpython3.10.so.1.0   /usr/lib/libpython3.8.so.1.0
```

jim-wilson, 2022-03-07:
You didn't mention where the riscv64-unknown-elf-gdb binary came from.  There is no guarantee that a binary compiled for one distro will run on a different distro, or even a different version of the same distro.  Eventually we will have distro level standards, but the RISC-V software ecosystem is too new for that.

TommyMurphyTM1234, 2022-09-27:
@KaminariOS - is this still a live issue for you or can it be closed? If the former then please provide more detailed information allowing the issue to be better understood and investigated.

KaminariOS, 2022-09-27:
> @KaminariOS - is this still a live issue for you or can it be closed? If the former then please provide more detailed information allowing the issue to be better understood and investigated.

No. I found a workaround.

TommyMurphyTM1234, 2022-09-27:
Ok thanks for the update.

closed 2022-09-27 by TommyMurphyTM1234