riscv-collab/riscv-gnu-toolchain #1035: errors when assembling

PPParticle, 2022-03-07:
Since I have noticed that `vl` could be directly read to a scalar register, I tried `csrr (a4), vl`. However, it can't work. Futhermore, the error messages seem the compiling toolchain couldn't assemble the code.

Error Messages:
```
$ riscv64-unknown-elf-as -march=rv64gcv --mabi=lp64d char.s -o assembly.o

char.s: Assembler messages:
char.s:9: Error: illegal operands `add (a4),t0,0'
```

Also, the version of my toolchain:
```
$ riscv64-unknown-elf-as --version
GNU assembler (GNU Binutils) 2.35
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `riscv64-unknown-elf'.
```
I really know what is going on. Is the version of my toolchain out-of-date?

kito-cheng, 2022-03-07:
Drop parentheses brackets: `csrr a4,vl`

PPParticle, 2022-03-07:
> Drop parentheses brackets: `csrr a4,vl`

Oh! THANKS! It works.

TommyMurphyTM1234, 2022-09-27:
Confusion about RISC-V assembly code syntax - Not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234