riscv-collab/riscv-gnu-toolchain #1038: Multilib: missing C++ headers for rv64gc/lp64d

cmuellner, 2022-03-15:
When building this repository with `--enable-multilib --enable-linux`, we get a Linux multilib toolchain.

Let's now create a simple C++ program `foo.cpp`:
```
#include <cstddef>
```

Let's now compile that for all valid combinations (rv32gc/ilp32, rv32gc/ilp32d, rv64gc/lp64, rv64/lp64d):
```
$ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv64gc -mabi=lp64 -c foo.cpp
$ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv32gc -mabi=ilp32 -c foo.cpp
$ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv32gc -mabi=ilp32d -c foo.cpp
$ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv64gc -mabi=lp64d -c foo.cpp
In file included from foo.cpp:1:
/opt/riscv-rvi/riscv64-unknown-linux-gnu/include/c++/11.1.0/cstddef:49:10: fatal error: bits/c++config.h: No such file or directory
   49 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

We can see, that the combination rv64gc/lp64d is missing the header file `bits/c++config.h`.

Let's look it up:
```
$ find /opt/riscv-rvi/ -name c++config.h
/opt/riscv-rvi/riscv64-unknown-linux-gnu/include/c++/11.1.0/riscv64-unknown-linux-gnu/lib64/lp64/bits/c++config.h
/opt/riscv-rvi/riscv64-unknown-linux-gnu/include/c++/11.1.0/riscv64-unknown-linux-gnu/lib32/ilp32/bits/c++config.h
/opt/riscv-rvi/riscv64-unknown-linux-gnu/include/c++/11.1.0/riscv64-unknown-linux-gnu/lib32/ilp32d/bits/c++config.h
/opt/riscv-rvi/riscv64-unknown-linux-gnu/include/c++/11.1.0/riscv64-unknown-linux-gnu/bits/c++config.h
```

Looks like the header is there, but not installed into `lib64/lp64d`.
Let's see if that's the case for glibc headers:
```
$ ls /opt/riscv-rvi/sysroot/lib64/
lp64  lp64d
```

Further info:
* The C++ headers are installed as part of GCC stage2
* GCC stage2 is configured with `[...] --prefix=/opt/riscv-rvi [...] --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc`
* The issue can be reproduced with upstream GCC

Any thoughts on how to fix this issue? Do we need to fix this in GCC's multilib build scripts?

cmuellner, 2022-03-15:
I found a workaround:

```
$ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv64imafdc -mabi=lp64d -c foo.cpp
$ echo $?
0
```

So we need a string-match of the `march`-string not only a semantic-match.

kito-cheng, 2022-03-16:
I've improve the multi-lib selection last year, but that isn't merge yet since that, and I might send again once stage 1 is open.
https://patchwork.ozlabs.org/project/gcc/list/?series=262586

cmuellner, 2022-03-16:
Thanks for pointing to that!

TommyMurphyTM1234, 2022-11-10:
This is still extant with the latest sources (branch: master, commit: c63d1b66120f860e3f6b16019c77d38b918b44ec).

There seem to be two possibly related issues

1. Canonicalization of the arch string
2. Location of toolchain default headers (and libs?) versus location of multilib headers (and libs?)

When the toolchain is configured (as it is by default) for `rv64gc` this gets canonicalized to `rv64imafdc` (I'm sure that in the recent past it was `rv64imafdc_zicsr` or `rv64imafdc_zicsr_zifencei`?) as reported by `-dumpspecs`:
```
./riscv64-unknown-linux-gnu-gcc -dumpspecs
...
*multilib_defaults:
march=rv64imafdc mabi=lp64d
...
```
But when compiling code using the toolchain it does not understand that when `-march=rv64gc` is explicitly specified  that this is the same as `-rv64imafdc`. Note that if the toolchain default is not specified at all (and is inferred by the toolchain) or is explicitly specified in its canonicalized form (as @cmuellner did) then things work OK.

Secondly the toolchain default headers are installed into, say, `.../iscv64-unknown-linux-gnu/include/c++/12.2.0/` but not lib (also) into the multilib directory structure `.../riscv64-unknown-linux-gnu/include/c++/12.2.0/riscv64-unknown-linux-gnu/lib<XLEN>/<abi>`.

However I'm little the wiser as to what is (are) the appropriate change(s) here to address this. There seem to be wider issues with the handling of the canonicalized arch string that affect other things and which is in a state of flux at the moment?

TommyMurphyTM1234, 2022-11-10:
> I found a workaround:
> 
> ```
> $ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv64imafdc -mabi=lp64d -c foo.cpp
> $ echo $?
> 0
> ```
> 
> So we need a string-match of the `march`-string not only a semantic-match.

As mentioned below, 

> I found a workaround:
> 
> ```
> $ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -march=rv64imafdc -mabi=lp64d -c foo.cpp
> $ echo $?
> 0
> ```
> 
> So we need a string-match of the `march`-string not only a semantic-match.

Another workaround is to not explicitly specify the toolchain default arch.
```
$ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-g++ -c foo.cpp
```

TommyMurphyTM1234, 2023-10-15:
> When building this repository with `--enable-multilib --enable-linux`, we get a Linux multilib toolchain.
>
> Let's now create a simple C++ program `foo.cpp`:
>
> ```
> #include <cstddef>
> ```
>
> Let's now [...]

This issue no longer occurs using a build of the toolchain at the latest commit:

* https://github.com/riscv-collab/riscv-gnu-toolchain/commit/6b1324367b879a9b89437846827b48151b26b412

closed 2023-10-15 by TommyMurphyTM1234