riscv-collab/riscv-gnu-toolchain #1039: how build so file by a generic ELF/Newlib toolchain

licl14, 2022-03-21:
This is the RISC-V C and C++ cross-compiler. It supports two build modes: a generic ELF/Newlib toolchain and a more sophisticated Linux-ELF/glibc toolchain.

so  how build so file by a generic ELF/Newlib toolchain

just like gcc-arm-none-eabi can build so for  armv8-m   ,who,don't  run linux

jim-wilson, 2022-03-21:
There is no shared library support in the RISC-V newlib port, and hence no shared library support in the generic ELF embedded toolchain.  Some targets like ARM do have shared library support in the embedded elf toolchain, but most don't.  To get shared library support, someone needs to define a mechanism for embedded elf shared libraries to work, and then implement the necessary support, which will require newlib work, and probably also binutils and gcc work.  There needs to be some way to process relocations at run-time, and currently we don't have one (other than for UNIX targets).

jim-wilson, 2022-03-24:
That support only works for x86 linux targets.  For an embedded system, without an OS that can run the dynamic linker for you, you need a different way for shared libraries to work.  No one has defined such a system for RISC-V yet.  And the linux newlib support was dropped upstream because it hasn't worked in years.

```
commit 5a0ab4454b94b5d976d5398bd5147ea6173b313f
Author: Mike Frysinger <vapier@gentoo.org>
Date:   Mon Feb 7 04:57:29 2022 -0500

    newlib: punt sys/linux support

    This was only used by the i?86-pc-linux-gnu target which we've removed,
    and even though it's using a "sys/linux/" dir to make it sound like it
    only depends on the Linux kernel, it's actually tied to glibc APIs built
    on top of Linux.  Since the code relies on internal glibc APIs and has
    been broken for some time, punt it all.  If someone wants to bring it
    back, they can try and actually keep the Linux-vs-glibc APIs separate.
```

TommyMurphyTM1234, 2022-11-09:
Closing this on the basis that (a) shared libs on a bare metal/newlib target without any defined shared lib loading mechanism doesn't really make much sense and (b) the definition of such a bare metal shared lib loading mechanism is outside the scope of this particular project.

closed 2022-11-10 by TommyMurphyTM1234