riscv-collab/riscv-gnu-toolchain #1040: fatal: repository 'https://anongit.freedesktop.org/pixman/' not found

jerryhethatday, 2022-03-23:
There is something wrong when I update the summodules of rocketchip. It seems that something called "pixman" could not be properly found. I opened an issue at Rocketchip repo: https://github.com/chipsalliance/rocket-chip/issues/2951. However, I realize this should be a toolchain problem. 
Can anyone shed some light? Thanks
@palmer-dabbelt

cmuellner, 2022-03-23:
This repo does not have a direct dependency for pixman, but QEMU has one.
So what you are observing is an issue of the QEMU build.

Try the following:
```
apt-get install -y libpixman-1-dev
```

jerryhethatday, 2022-03-23:
@cmuellner after I install the libpixman-1-dev using apt-get and rerun the git sub module update --init --recursive， it still complains about the same error message.
Thanks for your reply

cmuellner, 2022-03-23:
Given that the [pixman submodule in QEMU has been dropped in 2017](https://git.qemu.org/?p=qemu.git;a=commit;h=c12b6d70e384c769ca372e15ffd19b3e9f563662) this issue is for sure irrelevant for this repo.

palmer-dabbelt, 2022-03-23:
Agreed.  Looks like the Rocket copy of riscv-gnu-toolchain hasn't been updated in 4 years.  I'm surprised there's no bigger breakages folks are seeing, but even for this updating is the way to go (or just moving to a distribution toolchain, to avoid these sorts of issues entirely).

jerryhethatday, 2022-03-24:
> Agreed. Looks like the Rocket copy of riscv-gnu-toolchain hasn't been updated in 4 years. I'm surprised there's no bigger breakages folks are seeing, but even for this updating is the way to go (or just moving to a distribution toolchain, to avoid these sorts of issues entirely).

@palmer-dabbelt Thanks for you replay. Do you mean I should directly clone the toolchain from `https://github.com/riscv/riscv-gnu-toolchain` and compile it? I am cloning i t now, but it takes a while. I wonder if there will be some incompatibility issue not to use the toolchain version that rocketchip repo provided? I am a rookie of the `make` stuff. And I want to know if I can still use the makefile in the rocketchip repo to build that project(like : 
 `cd emulator` , `make` and then `make -jN run-asm-tests`)
I am still struggling with runing the `git submodule update --init --recursive` under the old rocketchip toolchain repo in different network settings. But it kept complaining about `fatal: repository 'https://anongit.freedesktop.org/pixman/' not found`  I opened this in the broser, and it returns not found:[https://anongit.freedesktop.org/pixman/](https://anongit.freedesktop.org/pixman/)

jerryhethatday, 2022-03-27:
I tried many times cloning the tool chain following what the  rocket chip repo instructed. But it doesn't work, I am kind of sure that the tool chain submodule in rocketchip is outdated(pixman can not be found). 
I then take a detour using chipyard repo to set up the tool chain environment and RISCV environment variable is correctly set up by running the env.sh. Finally, the rockitchip repo works!
Thanks, @cmuellner @palmer-dabbelt 
I will close this.

closed 2022-03-27 by jerryhethatday

palmer-dabbelt, 2022-10-11:
The error is during the submodule fetch, not the build, so installing the
library likely won't do anything.  This sort of thing is usually a network
problem and will resolve itself, but sometimes it's a moved repository.
This worked yesterday for me, I'll check again when I get to the office.