riscv-collab/riscv-gnu-toolchain #1042: Make error for gcc

richunorman-1, 2022-03-29:
I have configured using the following script
```script
./configure --prefix=$RISCV_E --with-arch=rv64imac_zba_zbb_zbc_zbs_zicsr_zifencei --with-abi=lp64 --with-cmodel=medany --enable-multilib
```
Output:
```output
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed

```
Make error encountered
```script
make
```
error:
```error
if test -f /home/richu/Documents/riscv-gcc12-15-02/riscv-tools/riscv-gnu-toolchain/riscv-gcc/contrib/download_prerequisites && test "false" = "true"; then cd /home/richu/Documents/riscv-gcc12-15-02/riscv-tools/riscv-gnu-toolchain/riscv-gcc && ./contrib/download_prerequisites; fi
rm -rf stamps/build-gcc-newlib-stage1 build-gcc-newlib-stage1
mkdir build-gcc-newlib-stage1
cd build-gcc-newlib-stage1 && /home/richu/Documents/riscv-gcc12-15-02/riscv-tools/riscv-gnu-toolchain/riscv-gcc/configure \
	--target=riscv64-unknown-elf \
	 \
	--prefix=/home/richu/Documents/riscv-gcc12-15-02/riscv-install \
	--disable-shared \
	--disable-threads \
	--disable-tls \
	--enable-languages=c,c++ \
	--with-system-zlib \
	--with-newlib \
	--with-sysroot=/home/richu/Documents/riscv-gcc12-15-02/riscv-install/riscv64-unknown-elf \
	--disable-libmudflap \
	--disable-libssp \
	--disable-libquadmath \
	--disable-libgomp \
	--disable-nls \
	--disable-tm-clone-registry \
	--src=.././riscv-gcc \
	 \
	--enable-multilib \
	--with-abi=lp64 \
	--with-arch=rv64imac_zba_zbb_zbc_zbs_zicsr_zifencei \
	--with-tune=rocket \
	CFLAGS_FOR_TARGET="-Os   -mcmodel=medany" \
	CXXFLAGS_FOR_TARGET="-Os   -mcmodel=medany"
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... riscv64-unknown-elf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether ln works... yes
checking whether ln -s works... yes
checking for a sed that does not truncate output... /bin/sed
checking for gawk... /usr/bin/gawk
configure: error: building out of tree but .././riscv-gcc contains host-x86_64-pc-linux-gnu.
Use a pristine source tree when building in a separate tree
Makefile:531: recipe for target 'stamps/build-gcc-newlib-stage1' failed
make: *** [stamps/build-gcc-newlib-stage1] Error 1

```

kito-cheng, 2022-03-29:
The error message say your source tree are not clean and might be configured before, so maybe you could try to run `git clean -f -x` on `riscv-gcc` folder, but plz commit the change and add files you've add, otherwise those change might lose.

richunorman-1, 2022-03-29:
I have dropped the changes and tried `git clean -f -x` on `riscv-gcc`. But same error appears.

jim-wilson, 2022-03-29:
What does "git status --ignored" in the riscv-gcc dir say?  It should not report anything.  If it does, it needs to be removed.  I think you actually need "git clean -d -f -x" to also remove directories like the host-x86_64-pc-linux-gnu dir.

richunorman-1, 2022-03-30:
`git status --ignored` showed some files.Now it is clear. Thank you.

closed 2022-03-30 by richunorman-1