riscv-collab/riscv-gnu-toolchain #1043: Make error when compiled for zifencei

dingiso, 2022-03-30:
Because  the **riscv-pk** repo needs zifencei extension , so I want to build with it. My command is as following
 
```bash
 ./configure --prefix=$RISCV --with-arch=rv64gc_zifencei
```

and the error comes 

```bash
checking for AF_UNIX... yes
checking for AF_INET6... yes
checking if mkdir takes one argument... no
Traceback (most recent call last):
  File "../.././riscv-gcc/gcc/config/riscv/arch-canonicalize", line 114, in <module>
    print (arch_canonicalize(arg))
  File "../.././riscv-gcc/gcc/config/riscv/arch-canonicalize", line 90, in arch_canonicalize
    key=longext_sort))
  File "../.././riscv-gcc/gcc/config/riscv/arch-canonicalize", line 81, in longext_sort
    raise Exception("Unsupported extension `%s`" % exts)
Exception: Unsupported extension `zifencei`
--with-abi=lp64d is not supported for ISA
make[1]: *** [Makefile:4368: configure-gcc] Error 1
make[1]: Leaving directory '/home/luruibo/git/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:535: stamps/build-gcc-newlib-stage1] Error 2
```
It say that the zifencei is unsupported. Is it wrong with my options ? how can I build a riscv64-unknown-elf-gcc with zifencei extensions ? Thanks.

jim-wilson, 2022-03-30:
Note that the tools only recently added support for zifencei, and this support is only upstream, it isn't in riscv-gnu-toolchain yet.  Also note that rv64gc includes zicsr and zifencei so there is no need to specify zifencei here.  You only need to specify zicsr or zifencei if you use rv64imafdc, and you are using tools that are using the new scheme.

Anyways, don't use zifencei with riscv-gnu-toolchain.  It isn't necessary or supported yet.  riscv-gnu-toolchain is still using the old scheme where fence.i is part of the I base instruction set.

dingiso, 2022-03-30:
Thanks for the notification , maybe I should fix it by change to the upstream

closed 2022-03-31 by dingiso

nmeum, 2022-04-01:
> Anyways, don't use zifencei with riscv-gnu-toolchain. It isn't necessary or supported yet. riscv-gnu-toolchain is still using the old scheme where fence.i is part of the I base instruction set.

Why is that? How would I make it possible to compile code that requires `zicsr` (for example because it uses `csrr`) with both riscv-gnu-toolchain and the GCC upstream version? For the latter I would need to compile with `-march=rv32i_zicsr` but that wouldn't be recognized by the former toolchain?

jim-wilson, 2022-04-01:
In old versions of the ISA,  zifencei and zicsr are part of the base instruction set.  In new versions of the ISA, zifence and zicsr are separate extensions.  This was a non-backwards compatible ISA change.  The ISA changed a few years ago, but the compiler support is changing only now.  riscv-gnu-toolchain supports the old ISA definition that includes the csr instructions and fence.i in the base instruction set, so you don't need to specify them, and you can't, because they aren't supported yet.

In both the old and new versions of the ISA, rv64gc includes the csr instructions and fence.i.  This is because in the new ISA, rv64gc expands to rv64imafdc_zicsr_zifencei.  In the old ISA, rv64gc expands to rv64imafdc, and the I base includes the csr and fence.i instructions.  So in both the old and new ISA versions, rv64gc always includes the csr instructions and fence.i.

This is a complicated and confusing situation here because of the non-backwards compatible ISA change.  But the simple solution is to use rv64gc whenever possible, and don't try to specify zicsr or zifencei unless you get an error that requires adding them.

If you need something that works with both old and new toolchains, and can't use 'g', then you may be screwed.  Complain to the ISA committee.  They are the ones that created the problem with an incompatible change to the ISA.  There are a few ways to work around the problem.  You can specify an explicit ISA version.  I.e. if you have a new toolchain, and specify an old ISA version, it will accept the csr instructions and fencei.i as part of the I base, because they were in the old ISA version.  Another solution is to check compiler version, and use different -march options for different compiler versions.

In the long term, the old ISA scheme will be obsoleted, and there won't be a problem anymore, but it will be a number of years until we reach this point.