riscv-collab/riscv-gnu-toolchain #1044: System Call Table Confusion

yourfavouriteuncle, 2022-04-01:
Hello,

I'm trying to write a user mode emulator for a risc-v binary. In several different websites [1] I see the system call numbers being very different from the ones listed here [2] Can you tell me why the Linux system call numbers here are different between risc-v and every other platform?

[1] https://chromium.googlesource.com/chromiumos/docs/+/HEAD/constants/syscalls.md
[2] https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/linux-headers/include/asm-generic/unistd.h

kito-cheng, 2022-04-01:
@yourfavouriteuncle I would suggest you can drop a mail to linux-riscv@lists.infradead.org, that's RISC-V specific mailing list for linux kernel.

cmuellner, 2022-04-01:
Even in your provided link, there is no common numbering of syscalls.

E.g.:
`0`...`read` on x86-64, `restart_syscall` on x86/arm32, `io_setup` on aarch64
`1`...`write` on x86-64, `exit` on x86/arm32, `io_destroy` on aarch64
`2`...`open` on x86-64, `fork` on x86/arm32, `io_submit` on aarch64

Syscall number assignments have always been an architecture/ABI-specific thing.
Differences between architectures/ABIs have always been there.
So I don't see where the RISC-V port would break a convention.

yourfavouriteuncle, 2022-04-01:
Yup, @cmuellner is right, after a bit more digging I found a nice web page to give me a better overview [1]. Thanks!

[1] https://marcin.juszkiewicz.com.pl/download/tables/syscalls.html

closed 2022-04-01 by yourfavouriteuncle