riscv-collab/riscv-gnu-toolchain #1046: undefined reference to `__wrap_puts'

TONG369, 2022-04-03:
hi~

When I use   riscv64-unknown-elf  the following problems occur, how to resolve it ?

[image]

jim-wilson, 2022-04-04:
This is what happens when you use the linker option --wrap=puts.  Calls to puts get replaced with calls to __wrap_puts, and then the user (or library) is expected to provice __wrap_puts which can emulate puts, or do some target specific processing before calling __real_puts which is what the original puts function is renamed to.

See the LD docs.  [https://sourceware.org/binutils/docs-2.38/ld/Options.html#Options](https://sourceware.org/binutils/docs-2.38/ld/Options.html#Options) and the --wrap docs are about 3/4 of the way down this page.

The real problem here isn't a toolchain problem.  It is a problem with the software you are compiling.  Maybe you need to check the docs for that program.  Or maybe ask someone familiar with it.  You didn't mention exactly what you are compiling, other than some random FreeRTOS demo, so we can't comment on that.

TONG369, 2022-04-05:
thanks your answer. the problem is solved

closed 2022-04-05 by TONG369