riscv-collab/riscv-gnu-toolchain #1047: Inconsistent function prototype for interrupt attribute

AdamWRichardson, 2022-04-05:
I have the following code:
```C++
__attribute__((interrupt)) void interruptHandler(void *) { .. }
```
When compiling this produces the error: `error: 'interrupt' function cannot have arguments` which is raised in the `gcc/config/riscv/riscv.c`. If I then change the code to
```C++
__attribute__((interrupt)) void interruptHandler() { ... }
```
I get the error `error: x86-64 'interrupt' attribute only applies to functions that have only a pointer parameter optionally followed by an integer parameter` which seems strange because it's clearly generated from some x86 compiler and I haven't been able to find where this error is actually raised in the code.

I'm compiling with g++ version 11.1.0 which comes straight from configuring and installing from this repo. The configure command used is `./configure --with-arch=rv32i --with-abi=ilp32`

jim-wilson, 2022-04-06:
Interrupt attributes are very target specific, and there should be no assumption that anything that works for one target will work for another.  Functions declared with interrupt attributes normally take no arguments, because there is no obvious way to pass an argument via a hardware interrupt.  It looks like a rather odd x86 feature that it requires arguments, might be some obscure feature of the x86 hardware that requires this.  If you want to know why, you would have to ask someone that knows how x86 interrupts work.  This is a RISC-V git, so not the right place for x86 questions.

See the gcc docs for info on how interrupt attributes work for your hardware target.

AdamWRichardson, 2022-04-06:
Hi @jim-wilson firstly thanks for the reply!

I'm raising this issue here because these are the errors I get when compiling for RISC-V using this toolchain which is why it's strange to see any x86 errors anyway.

It doesn't seem possible to use the interrupt attribute for RISC-V as it is currently documented in the GCC docs.

AdamWRichardson, 2022-04-07:
Thanks @jim-wilson, the problem was actually with clang-tidy which I had enabled for the build which was building for my host machine, hence the x86 errors.

closed 2022-04-07 by AdamWRichardson