riscv-collab/riscv-gnu-toolchain #1048: Duplicate `ret` instruction at end of function

xarantolus, 2022-04-07:
Hi, I'm compiling the following C function (in `fib.c`):

```c
int fib(int n)
{
    if (n <= 1) {
        return n;
    }

    int prev = 0;
    int current = 1;

    int tmp;
    for (int i = 2; i <= n; i++) {
        tmp = current;
        current += prev;
        prev = tmp;
    }
    return current;
}
```

Using the following commands:

For normal RISC-V: 

    /opt/riscv/bin/riscv64-unknown-linux-gnu-gcc -Wall -nostdlib -static -mabi=lp64f -c -Os -march=rv64g fib.c -o fib

And for RISC-V Compressed:

    /opt/riscv/bin/riscv64-unknown-linux-gnu-gcc -Wall -nostdlib -static -mabi=lp64f -c -Os -march=rv64gc fib.c -o fib_compressed
    

Then I disassembled both files using `/opt/riscv/bin/riscv64-unknown-linux-gnu-gdb ./fib_compressed -batch -ex 'disassemble/r fib'` (it also reproduces with `riscv64-unknown-linux-gnu-objdump`).
In both cases, the function ends with *two* `ret` instructions at the end:

    Disassembly for RISC-V (normal, with -Os):
    Dump of assembler code for function fib:
       0x0000000000000000 <+0>:	13 07 10 00	li	a4,1
       0x0000000000000004 <+4>:	93 07 05 00	mv	a5,a0
       0x0000000000000008 <+8>:	63 54 a7 02	bge	a4,a0,0x30 <fib+48>
       0x000000000000000c <+12>:	13 07 20 00	li	a4,2
       0x0000000000000010 <+16>:	13 05 10 00	li	a0,1
       0x0000000000000014 <+20>:	93 06 00 00	li	a3,0
       0x0000000000000018 <+24>:	13 06 05 00	mv	a2,a0
       0x000000000000001c <+28>:	1b 07 17 00	addiw	a4,a4,1
       0x0000000000000020 <+32>:	3b 85 a6 00	addw	a0,a3,a0
       0x0000000000000024 <+36>:	93 06 06 00	mv	a3,a2
       0x0000000000000028 <+40>:	e3 d8 e7 fe	bge	a5,a4,0x18 <fib+24>
       0x000000000000002c <+44>:	67 80 00 00	ret
       0x0000000000000030 <+48>:	67 80 00 00	ret
    End of assembler dump.

And the compressed version:

    Disassembly for RISC-V (compressed, with -Os):
    Dump of assembler code for function fib:
       0x0000000000000000 <+0>:	05 47	li	a4,1
       0x0000000000000002 <+2>:	aa 87	mv	a5,a0
       0x0000000000000004 <+4>:	63 5c a7 00	bge	a4,a0,0x1c <fib+28>
       0x0000000000000008 <+8>:	09 47	li	a4,2
       0x000000000000000a <+10>:	05 45	li	a0,1
       0x000000000000000c <+12>:	81 46	li	a3,0
       0x000000000000000e <+14>:	2a 86	mv	a2,a0
       0x0000000000000010 <+16>:	05 27	addiw	a4,a4,1
       0x0000000000000012 <+18>:	35 9d	addw	a0,a0,a3
       0x0000000000000014 <+20>:	b2 86	mv	a3,a2
       0x0000000000000016 <+22>:	e3 dc e7 fe	bge	a5,a4,0xe <fib+14>
       0x000000000000001a <+26>:	82 80	ret
       0x000000000000001c <+28>:	82 80	ret
    End of assembler dump.

However, when I leave out the `-Os` option, it appears to only generate a single `ret` instruction (as expected). Without knowing anything about the inner workings of the compiler my unfounded guess is that the `bge` instruction generates a `ret` instruction to jump to at the end, without checking if one is already there; not sure though.

I installed the toolchain using the steps for Linux recommended in the README (running [this script](https://github.com/xarantolus/iitm-surveying-isas/blob/main/riscv/install.sh)). I'm on `Ubuntu 20.04.4 LTS x86_64`, `Kernel: 5.13.0-39-generic` (`neofetch` output), but it also reproduces in GitHub Actions, e.g. see [this run](https://github.com/xarantolus/iitm-surveying-isas/runs/5863159030?check_suite_focus=true) with [these results](https://github.com/xarantolus/iitm-surveying-isas/releases/tag/401f925) (where the files with `optimized` in their name were compiled with `-Os`, so the command line from above, and those without were compiled as above but without `-Os`).

Version info:

    $ /opt/riscv/bin/riscv64-unknown-linux-gnu-gcc --version
    riscv64-unknown-linux-gnu-gcc (g5964b5cd727) 11.1.0
    Copyright (C) 2021 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

jim-wilson, 2022-04-09:
As Andrew mentioned, this isn't a RISC-V problem, it is a gcc problem, so should be reported in the FSF GCC bugzilla rather than here.

The -Os option disables some optimizations that usually (but not always) increase code size.  Sometimes it accidentally disables an optimization that would have reduced code size.  But the optimization passes are complex, and the interactions between various optimization passes are even more complex, so getting perfect results is not always possible.  It may or may not be practical to fix this problem.

It isn't a middle end problem, it is a backend optimization issue.  With -Os, we end up with two paths to the function epilogue.  One has the return value in a0, and the other has the return value in a5.  So the end of the function is
  b L5
L4:
  mv a0, a5
L5:
  ret
Then there is a branch optimization pass that eliminates the branch by copying the return instruction giving
  ret
L4:
  mv a0, a5
  ret
Then there is a constant propagation pass that figures out that a0 and a5 happen to already contain the same value, changing this to
  ret
L4
  ret
and only now at the very end of optimization do we have two returns next to each other.

Running yet another branch optimization pass would probably fix this, but there is a trade off between making the compiler slower for everyone and making the compiled code slightly smaller for a few people, so it isn't obvious if this is a good idea or not.

We don't see the same problem with -O1 or -O2 because they make different optimization choices along the way, getting different code at the end of the function which doesn't trigger this problem.

aswaterman, 2022-04-11:
Interesting.  Thanks for the explanation, Jim.

TommyMurphyTM1234, 2022-09-27:
> As Andrew mentioned, this isn't a RISC-V problem, it is a gcc problem, so should be reported in the FSF GCC bugzilla rather than here.

TommyMurphyTM1234, 2022-09-28:
> > As Andrew mentioned, this isn't a RISC-V problem, it is a gcc problem, so should be reported in the FSF GCC bugzilla rather than here.

This issue is already logged in the upstream GCC Bugzilla bug tracker:

* https://gcc.gnu.org/bugzilla/show_bug.cgi?id=105034

closed 2022-09-28 by TommyMurphyTM1234