riscv-collab/riscv-gnu-toolchain #1049: make linux rv32i error

lji503, 2022-04-07:
I got an error. making a linux about rv32 toolchain

//error

--with-abi=ilp32d is not supported for ISA rv32i
make[1]: *** [Makefile:4352: configure-gcc] Error 1
make[1]: Leaving directory '/home/hyeona/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:368: stamps/build-gcc-linux-stage1] Error 2
-------------------------------------------------------------------------------------
//Input

$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain
$ cd riscv-gnu-toolchain
$ sudo apt-get update
$ sudo apt-get install autoconf automake autotools-dev curl python3 libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev
$ ./configure --prefix=/opt/riscv/rv32i --with-arch=rv32i --with-abi=ilp32d
$ sudo make linux 

How can I solve this error?
Please tell me how to solve the error or another way to install toolchain for rv32i

I installed ubuntu

stnolting, 2022-04-07:
For `ilp32d` I think you'll need to have architecture that supports float/doubles like `rv32id`.
For `rv32i`-only you have to use the `ilp32` ABI.

lji503, 2022-04-07:
so yon mean I need to change ilp32d->ilp32 for rv32i? and if i chaged ilp32d-> ilp32, Can I use double?

closed 2022-04-07 by lji503

stnolting, 2022-04-08:
> mips nios2 powerpc riscv glibc requires the A extension

Then you need to add `a` to the `march` string - provided the targeted processor supports the atomics ISA extension.