riscv-collab/riscv-gnu-toolchain #1050: How can I do ubuntu RV32I Toolchain Setup?

lji503, 2022-04-08:
I got an error. making a linux about RV2I toolchain

//error

--with-abi=ilp32d is not supported for ISA rv32i
make[1]: *** [Makefile:4352: configure-gcc] Error 1
make[1]: Leaving directory '/home/hyeona/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:368: stamps/build-gcc-linux-stage1] Error 2

//Input

$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain
$ cd riscv-gnu-toolchain
$ sudo apt-get update
$ sudo apt-get install autoconf automake autotools-dev curl python3 libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev
$ ./configure --prefix=/opt/riscv/rv32i --with-arch=rv32i --with-abi=ilp32d
$ sudo make linux

How can I solve this error?
Please tell me how to solve the error or another way to install toolchain for rv32i

I installed ubuntu, ㅑ want to make a linux RV2I toolchian setup

zhongjuzhe, 2022-04-08:
There are 2 problems when you build up your riscv gnu toolchain.
1:`rv32i` means only support integer, but your `abi` imply that you want to support `double`. 
So there is a question: do you want to support `double` in your application?
2:Linux can't run with `A` extension which means you should include `atomic` ISA in your Linux RISCV toolchain.

Finally, I would suggest you:
1.If you want to support `double` in non-linux toolchain: you should configure `--with-arch=rv32id --with-abi=ilp32d`
2.If you don't want to support `double` in non-linux toolchain: you should configure `--with-arch=rv32i --with-abi=ilp32`
3.If you want to support `double` in linux toolchain: you should configure `--with-arch=rv32iad --with-abi=ilp32d`
4.If you don't want to support `double`  in linux toolchain: you should configure `--with-arch=rv32ia --with-abi=ilp32`

lji503, 2022-04-08:
> 
I do your suggestion 3
so I changed my input, --with-arch=rv32iad --with-abi-ilp32 
-----------------modyfied input------------------
$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain
$ cd riscv-gnu-toolchain
$ sudo apt-get update
$ sudo apt-get install autoconf automake autotools-dev curl python3 libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev
$ ./configure --prefix=/opt/riscv/rv32i --with-arch=rv32iad --with-abi=ilp32d
$ sudo make linux

make: *** [Makefile:340: stamps/build-glibc-linux-rv32iad-ilp32d] Error 1

zhongjuzhe, 2022-04-08:
Maybe I am wrong about this question. I didn't try `--with-arch=rv32iad --with-abi=ilp32d`. 
When I search in google, I find this website https://www.sifive.com/blog/all-aboard-part-5-risc-v-multilib. Maybe it can help you.
It seems in your situation, you should use `-march=rv32imafd -mabi=ilp32d`.

TommyMurphyTM1234, 2022-09-27:
Incorrect configuration of the toolchain using incompatible arch and abi. Not an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234