riscv-collab/riscv-gnu-toolchain #1051: make -j8 can not build gcc

vgegok, 2022-04-08:
When I used "make -j8" to build toochain, I can not get riscvXX-unknown-elf-gcc/g++………………

But I build toolchain with "make", It will working.

This is a BUG？

cmuellner, 2022-04-08:
I can build with `make -j16`.
What is the error message that you see (probably you need to scroll up)?

vgegok, 2022-04-08:
( rm -f BLD-POTFILES-t BLD-POTFILES \
    && (sed -e '/^#/d' \
            -e '/^[ 	]*$/d' \
	    -e "s@.*@	../& \\\\@" < /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/BLD-POTFILES.in \
	| sed -e '$s/\\$//') > BLD-POTFILES-t \
    && chmod a-w BLD-POTFILES-t \
    && mv BLD-POTFILES-t BLD-POTFILES )
cd .. \
  && CONFIG_FILES=po/Makefile.in:po/Make-in \
     CONFIG_HEADERS= /bin/bash ./config.status
config.status: creating po/Makefile.in
config.status: executing depfiles commands
config.status: executing libtool commands
config.status: executing default-1 commands
config.status: executing default commands
make[4]: Nothing to be done for 'info'.
make[4]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd/po'
make[4]: Entering directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd'
make[4]: Nothing to be done for 'info-am'.
make[4]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd'
make  all-recursive
make[4]: Entering directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd'
Making all in doc
make[5]: Entering directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd/doc'
make[5]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd/doc'
Making all in po
make[5]: Entering directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd/po'
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/da | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/da.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/es | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/es.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/fi | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/fi.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/fr | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/fr.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/hr | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/hr.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/id | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/id.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/ja | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/ja.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/pt | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/pt.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/ro | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/ro.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/ru | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/ru.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/rw | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/rw.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/sr | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/sr.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/sv | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/sv.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/tr | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/tr.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/uk | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/uk.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/vi | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/vi.po
file=`echo /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/zh_CN | sed 's,.*/,,'`.gmo \
  && rm -f $file && PATH=../src:$PATH /usr/bin/msgfmt -o $file /opt/riscv-gnu-toolchain/riscv-binutils/bfd/po/zh_CN.po
make[5]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd/po'
make[5]: Entering directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/bfd'
rm -f bfd-tmp.h
cp bfd-in3.h bfd-tmp.h
rm -f tofiles
/bin/bash /opt/riscv-gnu-toolchain/riscv-binutils/bfd/../move-if-change bfd-tmp.h bfd.h
  GEN      elf32-riscv.c
  GEN      elf64-riscv.c
f=""; \
[... 1252 lines cut ...]
libtool: install: /usr/bin/install -c nm-new /opt/riscv32im/bin/riscv32-unknown-elf-nm
  /bin/bash ./libtool  --mode=install /usr/bin/install -c strip-new /opt/riscv32im/bin/riscv32-unknown-elf-strip
libtool: install: /usr/bin/install -c strip-new /opt/riscv32im/bin/riscv32-unknown-elf-strip
[... cut ...]
make[6]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/binutils'
make[5]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/binutils'
make[4]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/binutils'
make[3]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib/binutils'
make[2]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib'
make[1]: Leaving directory '/opt/riscv-gnu-toolchain/build-binutils-newlib'
mkdir -p stamps/ && touch stamps/build-binutils-newlib

cmuellner, 2022-04-08:
No error in this log.

eroom1966, 2022-04-27:
There is a dependancy race somewhere here
When I run the following:-

./configure --enable-multilib
make -j8 linux

I occasionally get a failure in making the target 'multi-do'

jim-wilson, 2022-04-27:
I use parallel builds often and haven't seen a race, but maybe I just haven't triggered it.  If there is a race, it would be a problem with upstream sources not with riscv-gnu-toolchain, and we may not be able to fix it here.  Since I can't reproduce the problem, there is probably nothing I can do about it.

TommyMurphyTM1234, 2022-09-27:
No reproducible test case provided so closing this issue.