riscv-collab/riscv-gnu-toolchain #1052: Zicsr support in building RV32GC with --enable-multilib

jokap11, 2022-04-11:
Hi all,

I build the newlib toolchain with the -march=RV32GC --enable-multilib option and I have some weird observations while using the riscv32-unknown-elf-gcc compiler. First of all I tried to test CSR instructions with the march= rv32g & mabi=ilp32 option and got an unrecognized opcode error. Afterwards i tried to run the CSR instruction with the march= rv32imc & mabi=ilp32 option and got the same error. My next step was to try the same CSR instruction with the  march= rv32gc & mabi=ilp32 option and I got an "cant link double float modules with soft-float modules". But after changing the ABI to ilp32f it worked. Is there additional information what is actually supported with the architecture description. For example RV32IMAC is according to my observation also supporting ZiCSR instructions, but where can I look this up?
BR,
Johannes

pz9115, 2022-04-11:
In the psABI doc defined the RISC-V ABI calling convention, you can check the section "Default ABIs" for more informantion.
https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc

Basically we treating G expand to imafd and implied zicsr and zifencei, and for [released RISC-V ISA version 2.2](https://github.com/riscv/riscv-isa-manual/releases/tag/riscv-user-2.2) or earlier version, zicsr and zifencei is included in the base ISA I.  That is the case you mentioned.

And for new release [ISA version 20191213](https://github.com/riscv/riscv-isa-manual/releases/tag/Ratified-IMAFDQC), the zicsr and zifencei was cut from "i" extension. You should add "_zicsr" in march option to enable csr with the new ISA spec.

jim-wilson, 2022-04-11:
riscv-gnu-toolchain still implements the old ISA spec that includes the csr instructions in the base I instruction set.  We are in the process of switching to the new ISA spec.  That should happen in a few months.

You didn't give any testcases or show any errors, so we can't comment on why you are having trouble.

--enable-multilib doesn't build every possible arch/abi combination.  There are too many of them to build.  We only build a small subset.  If you choose an arch/abi combination that wasn't built, you will get linker errors.  You can see the available subset built by default by running gcc with --print-multi-lib.  If you want more multilibs than built by default, you can expand the multilib list.  You can use --with-multilib-generator= when configuring, or directly modify the gcc/config/riscv/t-elf-multilib file.

jim-wilson, 2022-04-11:
Turns out that riscv-gnu-toolchain has mismatched versions of binutils and gcc, where binutils is using the new scheme and hence needs zicsr but gcc is using the old scheme and does not accept zicsr.  This is unfortunate.  I'd suggest using a version of riscv-gnu-toolchain from before this change was made.  This might be fixable by telling binutils at configure time to use the old scheme.  See also #1053.

jokap11, 2022-04-12:
Hello @jim-wilson  @pz9115,

thank you for your extensive answers. Could you suggest the latest working commit number.
BR,
Johannes

jim-wilson, 2022-04-12:
Kito just merged in a patch to fix this.  It sets a default ISA version at configure time so that binutils and gcc will be configured to use the same ISA.  See #1054.  For now, this is the old ISA version which includes csr instructions in the I instruction base.  So you can just update to get the latest version, rerun configure to get the new ISA version default, rebuild the toolchain, and your problem should be solved.

jokap11, 2022-04-13:
Hello all,

thanks for the help it is finally working. I just want to mention that configuring the compiler to the old 20190806 scheme is not working with the same problems of missing opcode (CSR). So if anybody has hardships just use the standard configuration (=2.2).
BR,
Johannes