riscv-collab/riscv-gnu-toolchain #1053: Error: unrecognized opcode csrw

kuoyaoming93, 2022-04-11:
Hi all,

I am trying to install the RISC-V GNU toolchain in order to compile for [WD SweRV-EL2 core](https://github.com/chipsalliance/Cores-SweRV-EL2).

I followed the guidelines to install the toolchain, opting for the multilib option. 
```
# Clone the repository
git clone https://github.com/riscv/riscv-gnu-toolchain --recursive
cd riscv-gnu-toolchain/

# Configure as multilib and make
./configure --prefix=/opt/riscv --enable-multilib
make
```

After that, when I execute the following command to see the supported archs:

```
riscv64-unknown-elf-gcc --print-multi-lib

.;
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
```

Then, I cloned the [SweRV-EL2](https://github.com/chipsalliance/Cores-SweRV-EL2) RISC-V core, and try to run the hello_world simulation, the following error appeared when I build hello_world.o:

[image]

Did I forget something in the configuration stage?

Thank you in advance!

jim-wilson, 2022-04-11:
In the old ISA spec, the csr instructions are part of the base I instruction set.  In the new ISA spec, the csr instructions are part of the zicsr extension.  If -march=rv32imac doesn't work, then try -march=rv32imac_zicsr.  Old toolchains won't accept zicsr though as it is new.  So you may need to experiment to see what options your version of the assembler needs.

You can get the assembler version by running "riscv64-unknown-elf-as --version".

Worst case, you might need different arch options for gcc and gas if you have mismatched versions of them.

jim-wilson, 2022-04-11:
Then I would suggest using a version of riscv-gnu-toolchain from before this change was made.

closed 2022-04-11 by kuoyaoming93

jim-wilson, 2022-04-12:
Kito merged in a patch to fix this, so the top of the riscv-gnu-toolchain git tree should handle this correctly now.