riscv-collab/riscv-gnu-toolchain #1055: Adding custom instruction to vector compiler

Mousavikia, 2022-04-17:
Hi I have installed vector version of gcc using following instructions:
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic
cd riscv-gnu-toolchain
git submodule update --init --recursive
mkdir build && cd build
../configure --prefix=/desired/installation/path
make
```
I know that you (@jim-wilson ) said this branch is not reliable and llvm is better for vector extension, but consider I want to use this (since I don't know how to do the following in llvm), How can I add custom vector instructions to this version? I have done it for normal compiler (Main branch) is this totally same or should I do some thing else?

zhongjuzhe, 2022-04-17:
The RVV GCC has the full implementation including intrinsic and auto-vectorization. And I am actively maintaining recently and cleaning up the codes, it will be pushed upstream GCC in the future. I would suggest you use the branch `riscv-gcc-rvv-next` in riscv-gcc branch and `riscv-binutils-2.38` in riscv-binutils to build up the whole toolchain.
The intrinsic supports are all listed in here:https://github.com/riscv-non-isa/rvv-intrinsic-doc/tree/new-policy-proposal/overloaded_intrinsic_funcs
To support auto-vectorization: `-mrvv` `-mriscv-vector-bits`.

BHbean, 2022-04-21:
Excuse me, could you please tell me if there exists any docs for building?
I want to build a riscv-toolchain that contains rvv intrinsic using `riscv-gcc-rvv-next`  branch in [riscv-gcc](https://github.com/riscv-collab/riscv-gcc) repo. However, I did not see any guidance in the repo. Besides, as I have no admin permission on the server, I could not use `texi2html` command to generate an HTML doc to read.
So should I just follow the steps on https://gcc.gnu.org/install/? Are there any differences when building this `riscv-gcc-rvv-next`  branch?
Thanks in advance!

zhongjuzhe, 2022-04-21:
@BHbean  To build the whole toolchain, I suggest you use this repo: https://github.com/riscv-collab/riscv-gnu-toolchain which already assemble binutils, gcc, binutils,....,etc. Follow the guideline of README file, you can build up the whole toolchain.
To support RVV feature, you just need to checkout `riscv-gcc-rvv-next` in riscv-gcc and `riscv-binutils-2.38` in riscv-binutils.
There is no difference when building the toolchain with `riscv-gcc-rvv-next` branch in riscv-gcc.

BHbean, 2022-04-21:
@zhongjuzhe I think I got u. I will explain my understanding as followed.
**Firstly,** I just get the toolchain sources using
```
git clone https://github.com/riscv/riscv-gnu-toolchain -b rvv-intrinsic
```
**Then** I just need to change the `.gitmodules` file for the following modules
```
[submodule "riscv-binutils"]
	path = riscv-binutils
	url = https://github.com/riscv-collab/riscv-binutils-gdb.git
	branch = riscv-binutils-2.38
[submodule "riscv-gcc"]
	path = riscv-gcc
	url = https://github.com/riscv-collab/riscv-gcc.git
	branch = riscv-gcc-rvv-next
```
and leave other modules the same.
**Finally,** I just need to follow the guideline of README file to finish my build.
Did I make it right?

zhongjuzhe, 2022-04-21:
@BHbean Yes, I think you are right. Don't forget to git pull after you change the `.gitmodules` file. Besides, to enable RVV feature, you should bring `v` into `--with-arch` compile option. For example `--with-arch=rv64gcv`.

BHbean, 2022-04-21:
@zhongjuzhe Thanks a lot! Your suggestions are really useful! I will try it out as soon as possible!

MahmouodMagdi, 2022-05-25:
@zhongjuzhe can you help me to build the toolchain with vector support please ?
I just need the right commands to write it on the terminal

zhongjuzhe, 2022-05-26:
> @zhongjuzhe can you help me to build the toolchain with vector support please ? I just need the right commands to write it on the terminal

./configure --prefix=/opt/riscv --with-arch=rv64gcv_zfh --with-abi=lp64d
make

MahmouodMagdi, 2022-05-26:
> 
> 
> 
> 
> 
> 
> 
> 
> > @zhongjuzhe can you help me to build the toolchain with vector support please ? I just need the right commands to write it on the terminal
> 
> 
> 
> ./configure --prefix=/opt/riscv --with-arch=rv64gcv_zfh --with-abi=lp64d
> 
> make

Thanks for your help ♥️

LiQi19970428, 2022-06-03:
@zhongjuzhe 
Hi I have installed vector version of gcc，but there is something wrong with auto-vectorization: -mrvv。
And I have also changed the .gitmodules file。Modules are as followed:
$ git submodule
-9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc
-85e0e3519655220688e757b9d5bfd314923548bd musl
 415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)
-553032db17440f8de011390e5a1cfddd13751b0b qemu
 20756b0fbe065a84710aa38f2457563b57546440 riscv-binutils (heads/riscv-binutils-2.38)
-4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu
+05f2bee1fab961c3dedd46af317dc3320886a731 riscv-gcc (heads/riscv-gcc-rvv-next)
 5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)

But it went wrong as followed:
riscv64-unknown-elf-gcc: error: unrecognized command line option '-mrvv'

What should I do?
Thanks.

zhongjuzhe, 2022-06-03:
I removed -mrvv compile option to enable auto-vectorization. auto-vectorization is enable by -free-vectorize which is common GCC compile option. If you use - O3，the autovectorization is enabled by default.---- Replied Message ***@***.***>Date06/03/2022 19:14 ***@***.***> ***@***.******@***.***>SubjectRe: [riscv-collab/riscv-gnu-toolchain] Adding custom instruction to vector compiler (Issue #1055)

@zhongjuzhe

Hi I have installed vector version of gcc，but there is something wrong with auto-vectorization: -mrvv。

And I have also changed the .gitmodules file。Modules are as followed:

$ git submodule

-9826b03b747b841f5fc6de2054bf1ef3f5c4bdf3 glibc

-85e0e3519655220688e757b9d5bfd314923548bd musl

415fdd4279b85eeec9d54775ce13c5c412451e08 newlib (newlib-4.1.0)

-553032db17440f8de011390e5a1cfddd13751b0b qemu

20756b0fbe065a84710aa38f2457563b57546440 riscv-binutils (heads/riscv-binutils-2.38)

-4ea498a8e1fafeb568530d84db1880066478c86b riscv-dejagnu

+05f2bee1fab961c3dedd46af317dc3320886a731 riscv-gcc (heads/riscv-gcc-rvv-next)

5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)

But it went wrong as followed:

riscv64-unknown-elf-gcc: error: unrecognized command line option '-mrvv'

What should I do?

Thanks.

[

{

"target": "https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055#issuecomment-1145858821",

"url": "https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055#issuecomment-1145858821",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

LiQi19970428, 2022-06-06:
I used the examples you provided, but the results obtained did not seem to support RVV.The command is
riscv64-unknown-elf-gcc -ftree-vectorize -O3 -S auto.c
and the result is 
foo:
        beqz    a3,.L1
        slli    a3,a3,1
        add     a6,a2,a3
.L3:
        lb      a5,0(a0)
        lb      a3,0(a1)
        lhu     a4,0(a2)
        addi    a2,a2,2
        mulw    a5,a5,a3
        addi    a0,a0,1
        addi    a1,a1,1
        addw    a5,a5,a4
        sh      a5,-2(a2)
        bne     a6,a2,.L3
.L1:
        ret

zhongjuzhe, 2022-06-06:
> riscv-gcc (heads/riscv-gcc-rvv-next)
> 5da071ef0965b8054310d8dde9975037b0467311 riscv-gdb (remotes/origin/fsf-gdb-10.1-with-sim)

It's weird. Can you use `riscv64-unknown-elf-gcc -v` and show me the message in your terminal?

zhongjuzhe, 2022-06-06:
> 
You can see the last sentence: `gcc version 9.3.0 ()`. You are using the wrong GCC. The RVV GCC should be gcc version 12.

LiQi19970428, 2022-06-07:
Thanks.
I used git log and the last commit is you.

LiQi19970428, 2022-06-07:
It's on /usr/bin/riscv64-unknown-elf-gcc,and according to the date of the file, it is indeed an old version. So is compilation failed?

zhongjuzhe, 2022-06-07:
Use `/home/liqi/riscv/bin/riscv64-unknown-elf-gcc` -O3 -S auto.c.

***@***.***
 
From: Late
Date: 2022-06-07 17:32
To: riscv-collab/riscv-gnu-toolchain
CC: zhongjuzhe; Mention
Subject: Re: [riscv-collab/riscv-gnu-toolchain] Adding custom instruction to vector compiler (Issue #1055)
It's on /usr/bin/riscv64-unknown-elf-gcc,and according to the date of the file, it is indeed an old version. So is compilation failed?

LiQi19970428, 2022-06-07:
Thank you  for your help. Sorry to take up your precious time.

LiQi19970428, 2022-06-09:
Yes，thanks。

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2

closed 2022-11-10 by TommyMurphyTM1234