riscv-collab/riscv-gnu-toolchain #1056: Escaping Special Symbols in Linker Script

SaiVK, 2022-04-17:
Hello Everyone
  Is there any way to have special symbols like **_<_**, **_>_**, **_@_** as part of section names in the linker script? 

Thanking you
Sai

jim-wilson, 2022-04-17:
In theory quotes will work, but if you do this other things may break.

```
rohan:2129$ cat tmp.s
	.section "foo><@"
	nop
rohan:2130$ cat tmp.ld
SECTIONS
{
  . = 0x10000;
  "foo><@" : { *("foo><@") }
}
rohan:2131$ riscv64-unknown-elf-gcc tmp.s -Ttmp.ld -nostartfiles
rohan:2132$ riscv64-unknown-elf-objdump -h a.out

a.out:     file format elf64-littleriscv

Sections:
Idx Name          Size      VMA               LMA               File off  Algn
  0 foo><@        00000002  0000000000010000  0000000000010000  00000040  2**0
                  CONTENTS, READONLY
  1 .riscv.attributes 00000033  0000000000000000  0000000000000000  00000042  2**0
                  CONTENTS, READONLY
rohan:2133$ 
```

SaiVK, 2022-04-18:
Thanks @jim-wilson. What might break? If possible can you give an example? 
Is there any way to escape these symbols without breaking things?

Thanks
Sai

jim-wilson, 2022-04-18:
There are a lot of different programs that deal with object files and binaries, simulators, debuggers, shells, etc, and I can't guarantee that all of them will correctly handle unusual characters in section names.  But quotes should work for GNU as and ld.

SaiVK, 2022-04-18:
I see. Thanks, @jim-wilson. 

-Sai

TommyMurphyTM1234, 2022-09-27:
GNU linker question, not a RISC-V GCC toolchain issue.
Closing as invalid.

closed 2022-09-27 by TommyMurphyTM1234