riscv-collab/riscv-gnu-toolchain #1064: Failed to build the Linux cross-compiler

NgSherly, 2022-04-25:
Both master and basic-vvv branch also failed to build the linux cross-compiler with the step below:

./configure --prefix=/opt/riscv
make linux

**Error log:**
checking for suffix of object files... configure: error: in `/home/test/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details
make[1]: *** [Makefile:12417: configure-target-libgcc] Error 1
make[1]: Leaving directory '/home/test/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:379: stamps/build-gcc-linux-stage1] Error 2

**The error msg on config.log:**
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
configure:2215: $? = 0
configure:2204: gcc -V >&5
gcc: error: unrecognized command line option '-V'
gcc: fatal error: no input files
compilation terminated.
configure:2215: $? = 1
configure:2204: gcc -qversion >&5
gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
gcc: fatal error: no input files
compilation terminated.

is there any dependency need to be installed to get rid of this error?

jim-wilson, 2022-04-25:
There is a config.log file in every dir where configure runs.  You need to look at the one in the dir where it failed.  The dir name is in the error message.

The errors you listed aren't errors.  configure is designed to work with multiple compilers.  It tries every possible option spelling until it finds the one that works with the current compiler.  So it is normal to see it try an option or two that fails and then try one that works.  configure only fails if every option it tries fails, and this is a non-optional feature.

NgSherly, 2022-04-26:
Here is the config.log from **riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc**:

This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by GNU C Runtime Library configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ /home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/riscv-gcc/libgcc/configure --srcdir=../../.././riscv-gcc/libgcc --cache-file=./config.cache --with-cross-host=x86_64-pc-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-newlib --without-headers --disable-shared --disable-threads --with-system-zlib --enable-tls --disable-libatomic --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-bootstrap --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket --with-isa-spec=2.2 --enable-languages=c,lto --program-transform-name=s&^&riscv64-unknown-linux-gnu-& --disable-option-checking --with-target-subdir=riscv64-unknown-linux-gnu --build=x86_64-pc-linux-gnu --host=riscv64-unknown-linux-gnu --target=riscv64-unknown-linux-gnu

## --------- ##
## Platform. ##
## --------- ##

hostname = shuilein-NUC7i5DNHE
uname -m = x86_64
uname -r = 5.13.0-30-generic
uname -s = Linux
uname -v = #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/scripts/wrapper/sed
PATH: /opt/riscv/bin
PATH: /usr/local/sbin
PATH: /usr/local/bin
PATH: /usr/sbin
PATH: /usr/bin
PATH: /sbin
PATH: /bin
PATH: /snap/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:2079: creating cache ./config.cache
configure:2300: checking build system type
configure:2314: result: x86_64-pc-linux-gnu
configure:2334: checking host system type
configure:2347: result: riscv64-unknown-linux-gnu
configure:2450: checking for --enable-version-specific-runtime-libs
configure:2463: result: no
configure:2534: checking for a BSD-compatible install
configure:2602: result: /usr/bin/install -c
configure:2618: checking for gawk
configure:2645: result: /usr/bin/gawk
configure:2752: checking for riscv64-unknown-linux-gnu-ar
configure:2779: result: riscv64-unknown-linux-gnu-ar
configure:2844: checking for riscv64-unknown-linux-gnu-lipo
configure:2871: result: riscv64-unknown-linux-gnu-lipo
configure:2936: checking for riscv64-unknown-linux-gnu-nm
configure:2963: result: /home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/nm
configure:3028: checking for riscv64-unknown-linux-gnu-ranlib
configure:3055: result: riscv64-unknown-linux-gnu-ranlib
configure:3120: checking for riscv64-unknown-linux-gnu-strip
configure:3147: result: riscv64-unknown-linux-gnu-strip
configure:3209: checking whether ln -s works
configure:3213: result: yes
configure:3230: checking for riscv64-unknown-linux-gnu-gcc
configure:3257: result:  /home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv64-unknown-linux-gnu/bin/ -B/opt/riscv/riscv64-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv64-unknown-linux-gnu/include -isystem /opt/riscv/riscv64-unknown-linux-gnu/sys-include   
[... 196 lines cut ...]
target_subdir='riscv64-unknown-linux-gnu'
thread_header=''
tm_defines=''
tm_file=''
tmake_file=''
toolexecdir='$(exec_prefix)/$(target_noncanonical)'
toolexeclibdir='$(toolexecdir)/lib'
unwind_header=''
use_tm_clone_registry=''
vis_hide=''
with_aix_soname='aix'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "GNU C Runtime Library"
#define PACKAGE_TARNAME "libgcc"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "GNU C Runtime Library 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL "http://www.gnu.org/software/libgcc/"

configure: exit 1

jim-wilson, 2022-04-27:
This is the important info
```
configure:3800: /home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/xgcc -B/home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/ -B/opt/riscv/riscv64-unknown-linux-gnu/bin/ -B/opt/riscv/riscv64-unknown-linux-gnu/lib/ -isystem /opt/riscv/riscv64-unknown-linux-gnu/include -isystem /opt/riscv/riscv64-unknown-linux-gnu/sys-include -c -O2 -mcmodel=medlow conftest.c >&5
/home/shuilein/ssd/riscv/test/riscv-gnu-toolchain/build-gcc-linux-stage1/./gcc/as: line 106: exec: -m: invalid option
```
This is the last error before configure gives up and prints the cache variables, which is info that may be needed for tracing.

So you need to figure out why the assembler is broken.  The error suggests that you might be accidentally using the wrong assembler.  You can try running this command by hand adding the -v (aka verbose) option to see what the gcc driver is doing.  One question is what exactly that stray -m option is.  I would guess a -march or -mabi option which should be OK.   And if you were accidentally using an x86-64 as instead of a RISC-V as, I would expect a different error message.  I think the real problem is with the as shell script.  This is created at configure time.  It should be pointing at the RISC-V as in the build tree.  If there are problems with your PATH environment variable that can cause assembler issues, but I don't recall it failing exactly this way.  You can try running that as shell script by hand to see what it does.  Or you can try looking at it with more/less/emacs/vi/etc to see what it is doing, and make sure it is using the right assembler.  If the gcc as shell script is using the right assembler, then you may have a build error with your assembler.  Try testing your assembler binary in the install tree to see if it is working correctly.

NgSherly, 2022-05-06:
Somehow it is managed to build successfully now. I did make any changes but just re-clone the repo. Thanks for the help.

closed 2022-05-06 by NgSherly