riscv-collab/riscv-gnu-toolchain #1067: 【Strange question】cannot find gmp mpfr mpc, but I have installed these

athurwo, 2022-04-29:
my os is fodoera, I have installed gmp mpfr mpc, but cannot find those. like below:

**checking for the correct version of mpfr.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 3.1.0+ and MPC 0.8.0+.**
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
https://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
**make: *** [Makefile:545: stamps/build-gcc-newlib-stage1] Error 1**

`git submodule update --remote 
 sudo dnf install autoconf automake python3 libmpc-devel mpfr-devel gmp-devel gawk  bison flex texinfo patchutils gcc gcc-c++ zlib-devel expat-devel
 ./configure --prefix=/opt/riscv-elf
 sudo make`

I have install libmpc-devel mpfr-devel gmp-devel successfully.

why cannot find them?

Please help me to resolve this question.
Thanks.

athurwo, 2022-04-29:
> 

Thanks for your reply. I have known where misunderstand was. Before I installed mpfr in my os: /opt/mpfr, but this should be installed in riscv-gnu-toolchain/riscv-gcc/, and I run the contrib/download_prerequisites, but generate the package of mpfr is bad, so I download mpfr from mpfr website， now this question is disappear, but a new question is appear like below:

checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for style of include used by make... GNU
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc -B/home/fso/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ -isystem /home/fso/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/targ-include -isystem /home/fso/riscv-gnu-toolchain/newlib/newlib/libc/include -B/home/fso/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/fso/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/libnosys -L/home/fso/riscv-gnu-toolchain/newlib/libgloss/riscv64   
checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in `/home/fso/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib':
**configure: error: cannot compute suffix of object files: cannot compile**
See `config.log' for more details
make[2]: *** [Makefile:8448: configure-target-newlib] Error 1
make[2]: Leaving directory '/home/fso/riscv-gnu-toolchain/build-newlib'
make[1]: *** [Makefile:879: all] Error 2
make[1]: Leaving directory '/home/fso/riscv-gnu-toolchain/build-newlib'
make: *** [Makefile:579: stamps/build-newlib] Error 2

In my install path /opt/riscv-elf/bin , have some commands like:
riscv64-unknown-elf-c++  riscv64-unknown-elf-g++  riscv64-unknown-elf-gcc-10.2.0  riscv64-unknown-elf-gcc-nm      riscv64-unknown-elf-gcov       riscv64-unknown-elf-gcov-tool
riscv64-unknown-elf-cpp  riscv64-unknown-elf-gcc  riscv64-unknown-elf-gcc-ar      riscv64-unknown-elf-gcc-ranlib  riscv64-unknown-elf-gcov-dump  riscv64-unknown-elf-lto-dump

This looks did not complete but soon, Please help me again.
Thanks.

jim-wilson, 2022-04-29:
You can find the error details in the config.log file in the dir where the build failed.  It looks like there is something wrong with the compiler you built.  Maybe the assembler is failing for instance.  You can try compiling a trivial program to see if the cross compiler is working.  You won't have libraries yet so you won't be able to use header files or link, but you should be able to produce an object file.  E.g. "touch tmp.c; riscv64-unknown-elf-gcc -c tmp.c" should work and produce a riscv64 object file tmp.o.

If you have too much trouble trying to build the software you could try downloading a pre-built toolchain.  Some linux distros have riscv64 cross compiler packages you can install, though usually you need a recent one to get the tools.  Most have only linux cross compilers, but debian/ubuntu have embedded elf cross compilers also.  You still haven't mentioned what distro and version you have.

TommyMurphyTM1234, 2022-09-27:
This is an issue with the user's toolchain build environment and not with the RISC-V GCC toolchain itself. Closing this as invalid.

closed 2022-09-27 by TommyMurphyTM1234

Kingxukai, 2024-10-09:
You can try with these commands below:

``` bash
# to get the path
$ brew --prefix gmp mpfr
/path/to/gmp
/path/to /mpfr

# check it if exsist
$ ls /path/to/gmp/lib /path/to/gmp/include
$ ls /path/to/mpfr/lib /path/to/mpfr/include

# and then set the environment variables
$ export LDFLAGS="-L/path/to/gmp/lib -L/path/to/mpfr/lib"
$ export CPPFLAGS="-I/path/to/gmp/include -I/path/to/mpfr/lib"

# compile
$ make xx
```