riscv-collab/riscv-gnu-toolchain #1068: GDB error : dwarf2_find_location_expression: Corrupted DWARF expression

eroom1966, 2022-05-04:
I have an issue that when building the riscv-gnu-toolchain for a Linux target
the gdb which is built cannot correctly debug the executables built by the toolchain.

I see a similar closed issue here
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/935

When hitting breakpoints I see error messages of the form

```
Breakpoint 1, __printf (format=<error reading variable: dwarf2_find_location_expression: Corrupted DWARF expression.>)
    at printf.c:32
```

Additionally I cannot perform a _where_ command as it does not seem to walk the stack correctly.
For example, I know that I have a recursive program with a stack of at least 5 calls deep, but I only see the bottom of the stack containing the error message, and the top of the stack.

When looking at my git clone repo after running the commands to build the toolchain, I notice I have 2 copies of gdb
riscv-gnu-toolchian/riscv-binutils/gdb
riscv-gnu-toolchian/riscv-gdb/gdb

Referring back to issue 935 mentioned above, the patch exists in one of these gdb versions (riscv-gnu-toolchian/riscv-binutils/gdb/dwarf2/loc.c), but not the other (riscv-gnu-toolchian/riscv-gdb/gdb/dwarf2/loc.c)

I presume the gdb source being used to build the riscv/Linux target gdb is riscv-gnu-toolchian/riscv-gdb, which has this bug.

My question is, what do I do to resolve this bug - I am unable to debug any code running on a Linux target, this seems like a very common requirement, so I am surprised others have not highlighted this, am I doing something wrong ?

Thx
Lee

herocbn, 2022-05-12:
I also  have this problem

eroom1966, 2022-05-12:
I think I managed to overcome this issue by removing the following flags from my compilation
-g -O0

and adding
-gdwarf-2

eroom1966, 2022-05-12:
Apologies, I meant the cross compilation to create the riscv application

herocbn, 2022-05-13:
I met this error in gdb-multiarch when setting up qemu-gdb , which runs nomally but cannot read varibles from gdb-multiarch

codefwer, 2022-08-13:
> I think I managed to overcome this issue by removing the following flags from my compilation -g -O0
> 
> and adding -gdwarf-2

hi,i met the same error ,t have been troubled by this error for many days.
My riscv-gcc and riscv-gdb's version are all 10.1 
Could you please tell me your detailed solution , i am sorry that i dont konw where to change the flags from my compilation

PilotAware, 2022-08-13:
When compiling your application, do not use -g, instead, use -gdwarf-2

codefwer, 2022-08-13:
> 

there are one command in my makefile:
`CFALGS = -Wall -Werror -O -fno-omit-frame-pointer -ggdb`
so i should change `-ggdb` to `-gdwarf-2`?

PilotAware, 2022-08-13:
Yes

codefwer, 2022-08-13:
> Yes

thanks for your help, it worked!!!!

TommyMurphyTM1234, 2022-09-27:
Incorrect use of debug info flags, not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234