riscv-collab/riscv-gnu-toolchain #1069: Base Address change causing Linker Error

AayushMisra, 2022-05-06:
Hie,
In spike.lds linker script, I modified the base address from 0x10010000 to 0x80000000.
```
OUTPUT_ARCH( "riscv" )

SECTIONS
{
  . = 0x80000000;
  .text : { *(.text) }
  .data : { *(.data) }
}

```
Running the following commands:
```
$ riscv64-unknown-elf-gcc -g -Og -o rot13-64.o -c rot13.c
$ riscv64-unknown-elf-gcc -g -Og -T spike.lds -nostartfiles -o rot13-64 rot13-64.o
```
The following error was encountered:
`/debugger/ocd_test/rot13.c:8:(.text+0x10): relocation truncated to fit: R_RISCV_HI20 against symbol 'wait' defined in .sdata section in rot13-64.o collect2: error: ld returned 1 exit status`
My toolchain is compiled using -mcmodel=medany
Modifying to other lower addresses works fine, can someone help me to understand this issue better,
And can someone direct me to a link to understand RISC-V linker better.

kito-cheng, 2022-05-06:
Actually that's code model issue, the simple and quickest solution is  re-compile with `-mcmodel=medany`, and this post has more explanation about code model: https://www.sifive.com/blog/all-aboard-part-4-risc-v-code-models

TommyMurphyTM1234, 2022-09-27:
Hi @AayushMisra - did @kito-cheng's comment help or is this still a live issue for you?

AayushMisra, 2022-09-27:
Sorry for being dormant a while, Was unable to find the root cause of the issue due to lack of time.
Closing the issue for now.

closed 2022-09-27 by AayushMisra