riscv-collab/riscv-gnu-toolchain #1070: About the load after called problem

victoryang00, 2022-05-06:
Here's my assembly code using the command `riscv64-unknown-elf-gcc -mabi=ilp32 -g -O0 -w -ggdb -static -march=rv32imac` to compile. The assembly code was generated by clang-15 latest github version using command `clang -mno-relax -no-integrated-as -O1 -w --target=riscv32-unknown-elf`

```assembly
	.text
	.attribute	4, 16
	.attribute	5, "rv32i2p0_m2p0_a2p0_c2p0"
	.file	"call_with_args.py"
	.globl	before_main
	.p2align	1
	.type	before_main,@function
before_main:
	.cfi_startproc
	addi	sp, sp, -16
	.cfi_def_cfa_offset 16
	sw	ra, 12(sp)
	.cfi_offset ra, -4
	#APP
	lui	a0, 8192
	mv	s11, a0
	#NO_APP
	call	heap.init@plt
	#APP
	mv	s10, gp
	add	s11, s11, s10
	li	s0, 0
	lw	ra, 12(sp)
	addi	sp, sp, 16
	ret
	#NO_APP
.Lfunc_end0:
	.size	before_main, .Lfunc_end0-before_main
	.cfi_endproc

	.globl	main
	.p2align	1
	.type	main,@function
main:
	.cfi_startproc
	addi	sp, sp, -16
	.cfi_def_cfa_offset 16
	sw	ra, 12(sp)
	.cfi_offset ra, -4
	#APP
	mv	s0, sp
	#NO_APP
	li	a0, 4
	call	($f)@plt
	call	makeint@plt
	call	print@plt
	#APP
	li	a7, 93	#exit system call
	ecall	
	#NO_APP
	lw	ra, 12(sp)
	addi	sp, sp, 16
	ret
.Lfunc_end1:
	.size	main, .Lfunc_end1-main
	.cfi_endproc

	.globl	$f
	.p2align	1
	.type	$f,@function
$f:
	.cfi_startproc
	addi	sp, sp, -16
	.cfi_def_cfa_offset 16
	sw	ra, 12(sp)
	.cfi_offset ra, -4
	sw	a0, 8(sp)
	lui	a0, %hi(const_9)
	addi	a0, a0, %lo(const_9)
	call	print@plt
	lw	a0, 8(sp)
	call	makeint@plt
	call	print@plt
	lw	a1, 8(sp)
	li	a0, 1
	call	($g)@plt
	lui	a0, %hi(const_10)
	addi	a0, a0, %lo(const_10)
	call	print@plt
	lw	a0, 8(sp)
	lw	ra, 12(sp)
	addi	sp, sp, 16
	ret
.Lfunc_end2:
	.size	$f, .Lfunc_end2-($f)
	.cfi_endproc

	.globl	$g
	.p2align	1
	.type	$g,@function
$g:
	.cfi_startproc
	addi	sp, sp, -16
	.cfi_def_cfa_offset 16
	sw	ra, 12(sp)
	.cfi_offset ra, -4
	sw	a1, 8(sp)
	sw	a0, 4(sp)
	lui	a0, %hi(const_11) //problem loading. loaded with const_11+16
	addi	a0, a0, %lo(const_11)
	call	print@plt
	lw	a0, 4(sp)
	call	makeint@plt
	call	print@plt
	lw	a0, 8(sp)
	call	makeint@plt
	call	print@plt
	lui	a0, %hi(const_12)
	addi	a0, a0, %lo(const_12)
	call	($h)@plt
	lui	a0, %hi(const_13)
	addi	a0, a0, %lo(const_13)
	call	print@plt
	lw	ra, 12(sp)
	addi	sp, sp, 16
	ret
.Lfunc_end3:
	.size	$g, .Lfunc_end3-($g)
	.cfi_endproc

	.globl	$h
	.p2align	1
	.type	$h,@function
$h:
	.cfi_startproc
	addi	sp, sp, -16
	.cfi_def_cfa_offset 16
	sw	ra, 12(sp)
	.cfi_offset ra, -4
	call	print@plt
	lw	ra, 12(sp)
	addi	sp, sp, 16
	ret
.Lfunc_end4:
	.size	$h, .Lfunc_end4-($h)
	.cfi_endproc

	.type	$object$prototype,@object
	.data
	.globl	$object$prototype
	.p2align	3
$object$prototype:
	.word	0
	.word	3
	.word	($object$dispatchTable)
	.size	$object$prototype, 12

	.type	$object$dispatchTable,@object
	.section	.sdata,"aw",@progbits
	.globl	$object$dispatchTable
	.p2align	3
$object$dispatchTable:
	.word	($object.__init__)
	.size	$object$dispatchTable, 4

	.type	$int$prototype,@object
	.data
	.globl	$int$prototype
	.p2align	3
$int$prototype:
	.word	1
	.word	4
	.word	($int$dispatchTable)
	.word	0
	.size	$int$prototype, 16

	.type	$int$dispatchTable,@object
	.section	.sdata,"aw",@progbits
	.globl	$int$dispatchTable
	.p2align	3
$int$dispatchTable:
	.word	($object.__init__)
	.size	$int$dispatchTable, 4

	.type	$bool$prototype,@object
	.data
	.globl	$bool$prototype
	.p2align	3
$bool$prototype:
	.word	2
	.word	4
	.word	($bool$dispatchTable)
	.byte	0
	.zero	3
	.size	$bool$prototype, 16

	.type	$bool$dispatchTable,@object
	.section	.sdata,"aw",@progbits
	.globl	$bool$dispatchTable
	.p2align	3
$bool$dispatchTable:
	.word	($object.__init__)
	.size	$bool$dispatchTable, 4

	.type	$str$prototype,@object
	.data
	.globl	$str$prototype
	.p2align	4
$str$prototype:
	.word	3
	.word	5
	.word	($str$dispatchTable)
	.word	0
	.word	0
	.size	$str$prototype, 20

	.type	$str$dispatchTable,@object
	.section	.sdata,"aw",@progbits
	.globl	$str$dispatchTable
	.p2align	3
$str$dispatchTable:
	.word	($object.__init__)
	.size	$str$dispatchTable, 4

	.type	$.list$prototype,@object
	.data
	.globl	$.list$prototype
	.p2align	4
$.list$prototype:
	.word	4294967295
	.word	5
	.zero	4
	.word	0
	.word	0
	.size	$.list$prototype, 20

	.type	const_9,@object
	.globl	const_9
	.p2align	4
const_9:
	.word	3
	.word	6
	.word	($str$dispatchTable)
	.word	7
	.word	.Lstr.const_9
	.size	const_9, 20

	.type	.Lstr.const_9,@object
	.section	.sdata,"aw",@progbits
.Lstr.const_9:
	.asciz	"start f"
	.size	.Lstr.const_9, 8

	.type	const_10,@object
	.data
	.globl	const_10
	.p2align	4
const_10:
	.word	3
	.word	6
	.word	($str$dispatchTable)
	.word	5
	.word	.Lstr.const_10
	.size	const_10, 20

	.type	.Lstr.const_10,@object
	.section	.sdata,"aw",@progbits
.Lstr.const_10:
	.asciz	"end f"
	.size	.Lstr.const_10, 6

	.type	const_11,@object
	.data
	.globl	const_11
	.p2align	4
const_11:
	.word	3
	.word	6
	.word	($str$dispatchTable)
	.word	7
	.word	.Lstr.const_11
	.size	const_11, 20

	.type	.Lstr.const_11,@object
	.section	.sdata,"aw",@progbits
.Lstr.const_11:
	.asciz	"start g"
	.size	.Lstr.const_11, 8

	.type	const_12,@object
	.data
	.globl	const_12
	.p2align	4
const_12:
	.word	3
	.word	5
	.word	($str$dispatchTable)
	.word	1
	.word	.Lstr.const_12
	.size	const_12, 20

	.type	.Lstr.const_12,@object
	.section	.sdata,"aw",@progbits
.Lstr.const_12:
	.asciz	"h"
	.size	.Lstr.const_12, 2

	.type	const_13,@object
	.data
	.globl	const_13
	.p2align	4
const_13:
	.word	3
	.word	6
	.word	($str$dispatchTable)
	.word	5
	.word	.Lstr.const_13
	.size	const_13, 20

	.type	.Lstr.const_13,@object
	.section	.sdata,"aw",@progbits
.Lstr.const_13:
	.asciz	"end g"
	.size	.Lstr.const_13, 6

	.section	.init_array,"aw",@init_array
	.p2align	2
	.word	before_main
	.section	".note.GNU-stack","",@progbits
```

The decompiled code is 
```assembly
00010180 <main>:
   10180:	1141                	addi	sp,sp,-16
   10182:	c606                	sw	ra,12(sp)
   10184:	840a                	mv	s0,sp
   10186:	4511                	li	a0,4
   10188:	2811                	jal	1019c <$f>
   1018a:	2169                	jal	10614 <makeint>
   1018c:	2389                	jal	106ce <print>
   1018e:	05d00893          	li	a7,93
   10192:	00000073          	ecall
   10196:	40b2                	lw	ra,12(sp)
   10198:	0141                	addi	sp,sp,16
   1019a:	8082                	ret

0001019c <$f>:
   1019c:	1141                	addi	sp,sp,-16
   1019e:	c606                	sw	ra,12(sp)
   101a0:	c42a                	sw	a0,8(sp)
   101a2:	87018513          	addi	a0,gp,-1936 # 15360 <const_9>
   101a6:	2325                	jal	106ce <print>
   101a8:	4522                	lw	a0,8(sp)
   101aa:	21ad                	jal	10614 <makeint>
   101ac:	230d                	jal	106ce <print>
   101ae:	45a2                	lw	a1,8(sp)
   101b0:	4505                	li	a0,1
   101b2:	2801                	jal	101c2 <$g>
   101b4:	89018513          	addi	a0,gp,-1904 # 15380 <const_10>
   101b8:	2b19                	jal	106ce <print>
   101ba:	4522                	lw	a0,8(sp)
   101bc:	40b2                	lw	ra,12(sp)
   101be:	0141                	addi	sp,sp,16
   101c0:	8082                	ret

000101c2 <$g>:
   101c2:	1141                	addi	sp,sp,-16
   101c4:	c606                	sw	ra,12(sp)
   101c6:	c42e                	sw	a1,8(sp)
   101c8:	c22a                	sw	a0,4(sp)
   101ca:	8b018513          	addi	a0,gp,-1872 # 153a0 <const_11> // it was mis calculated with 16 bit(size of const_11) deviation
   101ce:	2301                	jal	106ce <print>
   101d0:	4512                	lw	a0,4(sp)
   101d2:	2189                	jal	10614 <makeint>
   101d4:	29ed                	jal	106ce <print>
   101d6:	4522                	lw	a0,8(sp)
   101d8:	2935                	jal	10614 <makeint>
   101da:	29d5                	jal	106ce <print>
   101dc:	8d018513          	addi	a0,gp,-1840 # 153c0 <const_12>
   101e0:	2039                	jal	101ee <$h>
   101e2:	8f018513          	addi	a0,gp,-1808 # 153e0 <const_13>
   101e6:	21e5                	jal	106ce <print>
   101e8:	40b2                	lw	ra,12(sp)
   101ea:	0141                	addi	sp,sp,16
   101ec:	8082                	ret

000101ee <$h>:
   101ee:	1141                	addi	sp,sp,-16
   101f0:	c606                	sw	ra,12(sp)
   101f2:	29f1                	jal	106ce <print>
   101f4:	40b2                	lw	ra,12(sp)
   101f6:	0141                	addi	sp,sp,16
   101f8:	8082                	ret
	...
```
The load right after being called, the gp has 16bit deviation. Could this be quickly fixed by command line options?

victoryang00, 2022-05-06:
I'm using `riscv64-unknown-elf-gcc (GCC) 10.1.0` and newly compiled 11.1.0 did not help.

victoryang00, 2022-05-06:
I found sth related with https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/KnziiZtEJNo/m/M8Vfbw9UCgAJ

jim-wilson, 2022-05-08:
The URL you pointed at is a medany specific problem that should not be relevant here.

There isn't enough info for me to understand what is wrong.  Ideally, you would provide enough info to reproduce, but you just gave some assembler code (confusingly referring to it as both gcc and clang output) and some decompiled code, but no info on linker or linker options or linker script which is probably where the problem is.

To debug this, I would suggest as first step looking at the value in the gp register at run time, and the value of the __global_pointer$ symbol in the final executable.  These need to be the same value.  Then you need to look at the relocation for the failing instruction in the object file and make sure it is OK.

victoryang00, 2022-05-09:
Thanks for letting me know that $gp may changes. The bug occurs when calling alloc
```assembly
# Runtime support function alloc2 (realloc).
# Prototype address is in a0.
# Number of words to allocate is in a1.
  li a2, 4                                 # Word size in bytes
  mul a2, a1, a2                           # Calculate number of bytes to allocate
  add a2, gp, a2                           # Estimate where GP will move
  bgeu a2, s11, alloc2_15                  # Go to OOM handler if too large
  lw t0, 4(a0)                             # Get size of object in words
  mv t2, a0                                # Initialize src ptr
  mv t3, gp                                # Initialize dest ptr
alloc2_16:                                 # Copy-loop header
  lw t1, 0(t2)                             # Load next word from src
  sw t1, 0(t3)                             # Store next word to dest
  addi t2, t2, 4                           # Increment src
  addi t3, t3, 4                           # Increment dest
  addi t0, t0, -1                          # Decrement counter
  bne t0, zero, alloc2_16                  # Loop if more words left to copy
  mv a0, gp                                # Save new object's address to return
  sw a1, 4(a0)                             # Set size of new object in words
  mv gp, a2                                # Set next free slot in the heap
  jr ra                                    # Return to caller
alloc2_15:                                 # OOM handler
  li a0, 5                                 # Exit code for: Out of memory
  la a1, const_5                           # Load error message as str
  addi a1, a1, 16                          # Load address of attribute __str__
  j abort                                  # Abort
```

which changes the $gp, and I changed to `call sbrk` instead.

```assembly
# Runtime support function alloc2 (realloc).
# Prototype address is in a0.
# Number of words to allocate is in a1.
  addi sp, sp, -60
  sw ra, 28(sp)
  sw fp, 24(sp)
  addi fp, sp, 60
  sw a0, -4(fp)
  sw s1, -8(fp)
  sw a5, -12(fp)
  sw a4, -16(fp)
  sw a3, -20(fp)
  sw a2, -24(fp)
  sw a1, -28(fp)
  li a2, 4                                 # Word size in bytes
  mul a2, a1, a2                           # Calculate number of bytes to allocate
  add a2, gp, a2                           # Estimate where GP will move
  bgeu a2, s11, alloc2_15                  # Go to OOM handler if too large
  lw t0, -28(fp)                           # Get size of object in words
  slli a0, t0, 2
  call sbrk
  lw a1, -4(fp)
  lw t0, 0(a1)
  sw t0, 0(a0)
  lw t0, 1(a1)
  sw t0, 1(a0)
  lw t0, 2(a1)
  sw t0, 2(a0)
  lw s1, -8(fp)
  lw a5, -12(fp)
  lw a4, -16(fp)
  lw a3, -20(fp)
  lw a2, -24(fp)
  lw a1, -28(fp)
  lw ra, 28(sp)
  lw fp, 24(sp)
  addi sp, sp, 60
  jr ra                                    # Return to caller
alloc2_15:                                 # OOM handler
  li a0, 5                                 # Exit code for: Out of memory
  la a1, const_5                           # Load error message as str
  addi a1, a1, 16                          # Load address of attribute __str__
  j abort                                  # Abort
```

closed 2022-05-09 by victoryang00

kito-cheng, 2022-05-10:
@victoryang00 seems like your runtime library is customized, `gp` should not change at runtime except initialization, however your runtime library did the adjustment for `gp`, so I guess we are hard to answer any question about that due to we are lacking knowledge about your customize runtime :(

victoryang00, 2022-05-10:
> 

But I think the compiler‘s semantic of load of `.globl` should consider that $gp may change.

kito-cheng, 2022-05-10:
> But I think the compiler‘s semantic of load of .globl should consider that $gp may change.

No, the gp is computed by linker, so compiler never know that, ABI say gp won't changed during runtime, and linker rely on the fact to do optimization.

jim-wilson, 2022-05-10:
It looks like the library is using gp as a pointer to the end of heap.  That is inconsistent with linker relaxation which relies on the value of gp to point to the small data section.  If this is an embedded target, and they made a choice not to use linker relaxation, then they could have chosen to use gp for a different purpose.  Similarly, sometimes people use tp for a different purpose if they aren't using threads.  But code written this way needs to be clearly documented as using a variant ABI, so people don't run into problems when using it.