riscv-collab/riscv-gnu-toolchain #1072: Was unable to find riscv tool chain with "march=rv64imab" " mabi=lp64"

kumarbrk9, 2022-05-11:
Hi,
I'm trying to explore on bitmanip instructions and require riscv tool chain with configurations "march=rv64imab" " mabi=lp64".
Can you please suggest where can I find latest version with mentioned configurations. Also can you help me in answering if I can untar and use it directly or any installation procedure need to be followed?

Thanks in advance.

Regards,
Kumar

cmuellner, 2022-05-11:
There is no toolchain with `-march=rv64imab` as there is no single Bitmanip ISA extension.
You probably want the Zb* extensions (zba, zbb, zbc, zbs). Support for these extensions is upstream (and part of GCC 12.1).

jim-wilson, 2022-05-12:
We don't have zb* support in riscv-gnu-toolchain as we haven't upgraded to gcc-12 yet.  That release just happened a few days ago.  We will probably wait a month or so for early bugs to be caught before we risk upgrading to a new version.  You can try building riscv-gnu-toolchain using upstream gcc-12 sources.  In theory that should work.

We only provide sources.  There are some volunteers building packages.  Recent enough Debian/Ubuntu has an embedded riscv cross compiler package you can install.  Most Linux distros will have a linux riscv cross compiler package you can install, but it won't be multilib, so it will have only rv64gc/lp64d.  There is a package for Windows called xPack GNU RISC-V Embedded GCC.  But probably none of these use gcc-12 yet.

kumarbrk9, 2022-05-12:
Hi cmuellner & wilson,
Thanks a lot for your kind responses.

As wilson mentioned "can try building riscv-gnu-toolchain using upstream gcc-12 sources.", 
I've downloaded gcc 12.1 and is handy.
May I know what steps I need to follow next? 

- where exactly can I find basic riscv tool chain 
- how to install it and configure it with "march=rv64imab", " mabi=lp64"
- how can I link gcc 12.1?
Sorry for asking very basic things. It would be great if anyone can post the steps to be followed.

Thanks in advance.

Regards,
Kumar

cmuellner, 2022-05-12:
I propose that you use the build scripts of this repository and set the GCC sources to your local GCC 12 source directory (using the configure parameter `--with-gcc-src=/path/to/gcc`). E.g.:
```
./configure --prefix=/path/to/instdir --with-gcc-src=/path/to/gcc --with-arch=rv64gc --with-abi=lp64d
make -j$(nproc)
```

kumarbrk9, 2022-05-12:
Hi cmuellner,
Thanks for all your time in giving a response with the clear picture. Will try it. 

Regards,
Kumar

kumarbrk9, 2022-05-17:
Hi cmuellner,

- we have to build gcc, mpc, gmp, mpfr, and binutils all separately for riscv? or above suggestion which you mentioned should work?
- If need to build separately, May I know the steps other than "configure" & "make" you mentioned above?

Thanks in advance.

Regards,
Kumar

cmuellner, 2022-05-17:
This repository (`riscv-gnu-toolchain`) offers the possibility to override the location of the sources of the subprojects (GCC, Binutils, glibc, ...). This can be done by specifying the external sources at configure time (e.g. `--with-gcc-src=/path/to/gcc` or `--with-binutis-src=/path/to/binutils`). This is also documented in the README.md (at the end).

gmp and mpfr are dependencies and are expected to be installed on the host machine that builds the cross-toolchain. All dependencies are documented in the README.md (at the beginning).

So you can use this repository to build a toolchain with GCC 12.1 and Binutils 2.38 sources by downloading them separately and configuring where they can be found.

kumarbrk9, 2022-05-17:
Hi cmuellner,

Tried the way you mentioned. But still I face multiple errors which start with
Assembler messages:
Error: cannot find default versions of the ISA extension `b'

I actually pointed to GCC 12.1.0. But still I face this error.
following was my command.
./configure --prefix=riscvtools_path --with-arch=rv64imab --with-abi=lp64 --with-gcc-src=path_to_gcc-12.1.0 --with-binutils-src=path_to_binutils-2.38

cmuellner, 2022-05-17:
As mentioned in my first response: `rv64imab` does not exist. There is no `b` extension, but there are `zba`, `zbb`, `zbc`, and `zbs`.

kumarbrk9, 2022-06-21:
Hi cmuellner,

Have given a try with following command.
./configure --prefix=**path_to_instdir** --with-arch=rv64ima_zba_zbb_zbe_zbf_zbs_zicsr_zifencei --with-abi=lp64 --with-cmodel=medany --enable-multilib --with-gcc-src=**path_to_gcc-12.1.0** --with-binutils-src=**path_to_binutils-2.38**

But facing following errors. 
for i in argz/lib.a stdlib/lib.a ctype/lib.a search/lib.a stdio/lib.a string/lib.a signal/lib.a time/lib.a locale/lib.a reent/lib.a errno/lib.a misc/lib.a ssp/lib.a syscalls/lib.a machine/lib.a ; do \
riscv64-unknown-elf-ar x ../$i; \
done; \
riscv64-unknown-elf-ar rc ../libc.a *.o
riscv64-unknown-elf-ar: ../argz/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../stdlib/lib.a: No such file or directory
........
riscv64-unknown-elf-ar: ../syscalls/lib.a: No such file or directory
riscv64-unknown-elf-ar: ../machine/lib.a: No such file or directory
riscv64-unknown-elf-ar: *.o: No such file or directory
make[6]: *** [libc.a] Error 1

Can you please suggest what I'm missing out?

Thanks in advance

kito-cheng, 2022-06-21:
zbe and zbf isn't supported on upstream binutils 2.38

kumarbrk9, 2022-06-21:
Can you please suggest what version should I upgrade to, in order to get rid of errors?

closed 2022-07-06 by kumarbrk9