riscv-collab/riscv-gnu-toolchain #1075: Question about arithmetic shift right in RISC-V

zhongjuzhe, 2022-05-14:
Hi, When I am testing auto-vectorization of RVV GCC, I see there is a inconsistency between vector instructions and scalar instructions for dealing with arithmetic shift right operation in RISC-V.

Here is the code as follows(signbit-5.c):
```
#include <stdint.h>
#include <limits.h>
#include <stdio.h>

#ifndef N
#define N 65
#endif

#ifndef TYPE
#define TYPE int32_t
#endif

#ifndef DEBUG
#define DEBUG 1
#endif

#define BASE ((TYPE) -1 < 0 ? -126 : 4)

__attribute__ ((noinline, noipa))
void fun1(TYPE *x, int n)
{
    for (int i = 0; i < n; i++)
      x[i] = (-x[i]) >> 31;
}

__attribute__ ((noinline, noipa, optimize("O1")))
void fun2(TYPE *x, int n)
{
    for (int i = 0; i < n; i++)
      x[i] = (-x[i]) >> 31;
}

int main ()
{
  TYPE a[N];
  TYPE b[N];

  a[0] = INT_MIN;
  b[0] = INT_MIN;

  for (int i = 1; i < N; ++i)
    {
      a[i] = BASE + i * 13;
      b[i] = BASE + i * 13;
    }
  fun1 (a, N);
  fun2 (b, N);
  printf("result:%d,a:%d,b:%d\n",0,a[0],b[0]);
  for (int i = 0; i < N; ++i)
    {
      if (a[i] != b[i])
        printf("fail:%d,a:%d,b:%d\n",i,a[i],b[i]);
    }
  return 0;
}
```
ASM of riscv64-unknown-elf-gcc -mrvv -march=rv64gcv -O3 signbit-5.c
```
0000000000010270 <fun1>:
   10270:       02b05163                blez    a1,10292 <fun1+0x22>
   10274:       c2202773                csrr    a4,vlenb
   10278:       0505f7d7                vsetvli a5,a1,e32,m1,ta,mu
   1027c:       02056c07                vle32.v v24,(a0)
   10280:       0f804c57                vneg.v  v24,v24
   10284:       a78fbc57                vsra.vi v24,v24,31
   10288:       02056c27                vse32.v v24,(a0)
   1028c:       8d9d                    sub     a1,a1,a5
   1028e:       953a                    add     a0,a0,a4
   10290:       f5e5                    bnez    a1,10278 <fun1+0x8>
   10292:       8082                    ret

0000000000010294 <fun2>:
   10294:       00b05e63                blez    a1,102b0 <fun2+0x1c>
   10298:       872a                    mv      a4,a0
   1029a:       058a                    slli    a1,a1,0x2
   1029c:       952e                    add     a0,a0,a1
   1029e:       431c                    lw      a5,0(a4)
   102a0:       00f027b3                sgtz    a5,a5
   102a4:       40f007bb                negw    a5,a5
   102a8:       c31c                    sw      a5,0(a4)
   102aa:       0711                    addi    a4,a4,4
   102ac:       fea719e3                bne     a4,a0,1029e <fun2+0xa>
   102b0:       8082                    ret

```

This test compare the result between `func1` and `func2`. You can see assembly that `func1` is vectorized but `func2` is not. 

Here is the result of running the binary in the spike:
```
result:0,a:-1,b:0
fail:0,a:-1,b:0
```
The result of `func1` is **-1**` but the the result of `func2` is **0**. So this test fails finally during regression in dejagnu.

Also, I tried GCC of X86 with this `gcc -O3 signbit-5.c ` compile command and run with `./a.out`:
```
result:0,a:-1,b:-1
```
You can see the result of both `func1` and `func2` are **-1** for X86 GCC.  It is consistent.

Finally, I tried Clang of RISC-V with `clang --gcc-toolchain=/work/home/lding/apps/riscv64 --sysroot=/work/home/lding/apps/riscv64/sysroot --static --target=riscv64-unknow-linux-gnu -march=rv64gc  -O3 signbit-5.c` and run on the spike:

```
result:0,a:0,b:0
```
You can see the result of both `func1` and `func2` are **0** for RISC-V Clang.  It is consistent.

Can you help me with this question? I am confused about this question. Thanks~

cmuellner, 2022-11-15:
Hi @zhongjuzhe, sorry for the delayed reply!

Has this been resolved?

I just looked at the code and compared GCC/riscv64_rv64gc/upstream/qemu with GCC/x64_64/Fedora/native and Clang/x64_64/Fedora/native. All three resulted in `result:0,a:0,b:0` when using `-O3`.

So my findings are:
* I cannot reproduce your GCC x86 result
* The result of x86 and riscv64 match

I assume there might have been a GCC issue since you had the same unexpected result in both GCC runs with scalar code.
However, given this is not the case anymore, I conclude this issue got resolved.

Therefore I'm closing this ticket.
Feel free to reopen if I missed something.

closed 2022-11-15 by cmuellner

zhongjuzhe, 2022-11-15:
> Hi @zhongjuzhe, sorry for the delayed reply!
>
> Has this been resolved?
>
> I just looked at the code and compared GCC/riscv64_rv64gc/upstream/qemu with GCC/x64_64/Fedora/native and [...]

Thank. It's not fixed. But it's ok to close since it's not compiler issue. Instead, it's ISA issue.

cmuellner, 2022-11-15:
Has this been discussed elsewhere? If so, do you have a link where this has been discussed?

zhongjuzhe, 2022-11-15:
No, I didn't discuss it in other place. Maybe I will talk about it in the future if I care about it at that time.