riscv-collab/riscv-gnu-toolchain #1076: riscv64-oe-linux-gfortran: fatal error: cannot read spec file 'libgfortran.spec'

dozhier, 2022-05-15:
when i am trying to bulid an yocto project with openembedded-core & meta-riscv sources,i use  riscv64-oe-linux-gfortran as my gfortran complier,but it fails like this: riscv64-oe-linux-gfortran: fatal error: cannot read spec file 'libgfortran.spec': No such file or directory

i don't know why and looking for help,here is what i write in my .bb file.

[image]

jim-wilson, 2022-05-15:
The libgfortran.spec file is part of the gfortran install tree.  Sounds like something went wrong with your build.  Maybe you built the fortran compiler, but forgot to build the libgfortran library?  The libgfortran.spec file comes from the fortran library build.  In that case, you would also be missing the libgfortran.a and libgfortran.so files.  GNU fortran won't be usable without them.

TommyMurphyTM1234, 2022-09-27:
@dozhier - do you consider this to still be a live issue or can it be closed?

dozhier, 2022-09-28:
it can be closed.

TommyMurphyTM1234, 2022-09-28:
Thanks @dozhier .

closed 2022-09-28 by TommyMurphyTM1234