riscv-collab/riscv-gnu-toolchain #1077: _Float16 is not supported on this target

changduanyu, 2022-05-15:
Hi,

I was trying to build a FP16 application with the toolchain with "-march=rv32imfcv_zfh -mabi=ilp32f", and it gives "error: '_Float16' is not supported on this target".

It seems that _Float16 type is already defined,  what config should I use to make FP16 work?

Thanks!

jim-wilson, 2022-05-15:
What branch are you using?  There is no vector or zfh support on mainline.  There is experimental vector and zfh support on the riscv-gcc branch called riscv-gcc-rvv-next.  Using this probably requires using upstream binutils.  And because it is experimental, don't expect it to be well supported.  It will be a while before the vector support makes it upstream and then gets pulled down into riscv-gnu-toolchain.  The zfh support is unlikely to be handled separate from the vector work.

changduanyu, 2022-05-16:
Jim, thanks. I am using the main branch. But why the zfh support is coupled with vector extension, not a separated extension like single precision or double precision?

jim-wilson, 2022-05-18:
The vector extension support required the zfh support, so the zfh support was written when writing the vector support.  It is a separate extension, but it is only supported on the vector branch.  This will eventually be fixed.  But meanwhile, I don't know of any core that supports zfh without vector, so it isn't a critical problem that they are tied together.

changduanyu, 2022-05-22:
Thanks Jim, I will close this issue.

closed 2022-05-22 by changduanyu