riscv-collab/riscv-gnu-toolchain #1079: Does there have special option to enable bitmanip extension???

yuyang-ok, 2022-05-17:
[image]

https://github.com/riscv/riscv-bitmanip/blob/main/bitmanip/insns/andn.adoc

yuyang-ok, 2022-05-17:
I am using
~~~
$ riscv64-unknown-elf-as.exe   -version 
GNU assembler (GNU Binutils) 2.34
Copyright (C) 2020 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `riscv64-unknown-elf'.
~~~

kito-cheng, 2022-05-17:
You need `zbb` in your `-march=` option like `-march=rv64g_zbb`

yuyang-ok, 2022-05-17:
$ riscv64-unknown-elf-as.exe   -march=rv64g_zbb        a.s
Assembler messages:
Fatal error: -march=rv64g_zbb: unsupported ISA subset `z'

yuyang-ok, 2022-05-17:
Did I install unspported version of gcc tool chain??

yuyang-ok, 2022-05-17:
~~~
$ riscv64-unknown-elf-as.exe  -march=rv64g_zbb         a.s 
Assembler messages:
Fatal error: -march=rv64g_zbb: unsupported ISA subset `z'
~~~

yuyang-ok, 2022-05-17:
get unsupported  means???

closed 2022-05-17 by yuyang-ok

jim-wilson, 2022-05-18:
The zb* extensions are new, ratified last year, so you will only find them in the very latest binutils/gcc versions, released this year, which are not yet in riscv-gnu-toolchain.