riscv-collab/riscv-gnu-toolchain #1082: compilation error when tried to get .obj file from .as(Go) while using native assmbler to build

Sharanabasavaexaleap, 2022-05-26:
Hello,
We want to use native standalone assembler to build the .s file on HiFiveUnleashed (U540).

kernel version:
Linux Exaleap-Unmatched-Linux-1.3.0 5.11.1 riscv64 GNU/Linux
--------------
Go tool version:
o version go1.17.8 linux/riscv64
---------------
as -v:
GNU assembler version 2.35.1 (riscv64-oe-linux) using BFD version (GNU Binutils) 2.35.1
----------
The .s is created from "Go" compiler as given below.
Command used:
go tool compile -S sampple.go > sample.s
as sample.s -o samp.o

Errors file attached with this issue in below
=======
[sample.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/8779252/sample.txt)
[image]

Can you please help to fix this error?

kito-cheng, 2022-05-27:
Just tried `go tool compile`, the output seems not the expect format for GNU assembler, curious why you want to use GNU assembler instead builtin one (e.g. compile without `-S`)?

Sharanabasavaexaleap, 2022-06-24:
> `gccgo` and `go` are different tool, `gccgo` could generate ordinary assembly files, but apparently `go` is generating...some internal format which is not used for assembler.

Hi @kito-cheng  
Does "go" tool use "gccgo" for compilation? Or does it use a different compiler framework other than GNU framework?