riscv-collab/riscv-gnu-toolchain #1086: Build of newlib on Apple M1 fails because of outdated m4

ccattuto, 2022-06-06:
The build system prioritizes a binary named `gm4` over `m4`. On Apple M1 machines (tested with OSX 12.4) this resolves to either `/usr/bin/gm4` (`GNU M4 1.4.6` on OSX 12.4) or – when command-line tools are installed – to `/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/gm4` (still `GNU M4 1.4.6` on OSX 12.4). That old version of m4 does not support the `--gnu` option and building newlib fails. Homebrew carries a sufficiently recent version of m4 (`m4 (GNU M4) 1.4.19` as of today) but it is named `m4`, and the build process priorities the older `gm4` over it.

Either dropping `--gnu` on invoking `gm4` or allowing the user to override the default priorities for the m4 binary to be used seems necessary to correctly build newlib on Apple M1 machines.

cmuellner, 2022-09-29:
I just had a look at this and you are right, m4 in homebrew is [keg-only](https://docs.brew.sh/FAQ#what-does-keg-only-mean).

Maybe the following command helps: `export PATH="/opt/homebrew/opt/m4/bin:$PATH"`.

ccattuto, 2022-10-15:
Not really, because the build system finds a `gm4` first, and uses that.

TommyMurphyTM1234, 2022-11-12:
https://github.com/riscv-collab/riscv-gnu-toolchain/issues/800#issuecomment-1312187278

closed 2022-11-12 by TommyMurphyTM1234