riscv-collab/riscv-gnu-toolchain #1087: pk32 build error

Tong-Lin, 2022-06-07:
Run the following command:
```
./configure --prefix=/workspace/tonglin/release/ --enable-multilib  --with-sim=spike
make newlib buid-sim
```
Then, got following error:
```
/workspace/tonglin/release/riscv-toolchain/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /workspace/tonglin/release/riscv-toolchain/bin/../lib/gcc/riscv64-unknown-elf/11.1.0/libgcc.a(div.o)
riscv64-unknown-elf-objcopy -S -O binary --change-addresses -0x80000000 bbl bbl.bin
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:336: pk] Error 1
make[1]: Leaving directory '/workspace/tonglin/projects/riscv-gnu-toolchain/build/build-pk32'
make: *** [Makefile:825: stamps/build-pk32] Error 2
```

Add **"--with-abi=ilp32d"** option to target "stamps/build-pk32" can fix it
```
stamps/build-pk32: $(PK_SRCDIR) stamps/build-gcc-newlib-stage2
	rm -rf $@ $(notdir $@)
	mkdir $(notdir $@)
	cd $(notdir $@) && $</configure \
		--prefix=$(INSTALL_DIR) \
		--host=$(NEWLIB_TUPLE) \
                --with-abi=ilp32d \
		--with-arch=rv32gc
	$(MAKE) -C $(notdir $@)
	cp $(notdir $@)/pk $(INSTALL_DIR)/$(NEWLIB_TUPLE)/bin/pk32
	mkdir -p $(dir $@)
	date > $@
```

Liaoshihua, 2022-08-09:
```
/work/liaoshihua/release/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: /work/liaoshihua/release/lib/gcc/riscv64-unknown-elf/11.1.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/work/liaoshihua/release/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /work/liaoshihua/release/lib/gcc/riscv64-unknown-elf/11.1.0/libgcc.a(div.o)
```
I reproduced the problem with `make newlib SIM=spike build-sim` . I think `ABI is incompatible with that of the selected emulation:  target emulation elf64-littleriscv does not match elf32-littleriscv`  is the more accurate reason

Incarnation-p-lee, 2022-08-23:
Meet the same issue when compile the binaries like `make build-sim SIM=spike` when ` --enable-multilib` enabled. Consider the below snapshot from README, not sure if it is expected or not.

May be the problem is that if you would like to setup both `multilib` and `spike(pk)`, you may need two different dir to ensure the succeed of the build process. 

[image]

Incarnation-p-lee, 2022-08-25:
Take look into it, it may related to riscv-pk repo itself for its' RV32 support. Look it only support `rv32i` arch for now. You can reproduce this easily from `riscv-pk` repo for above error message with `--with-arch=rv32gc`.

_Quote:
By default, 64-bit (RV64) versions of pk and bbl are built. To built 32-bit (RV32) versions, supply a --with-arch=rv32i flag to the configure command._

I prepared one PR #1121  here for reference.

TommyMurphyTM1234, 2022-10-05:
> Run the following command:
>
> ```
> ./configure --prefix=/workspace/tonglin/release/ --enable-multilib  --with-sim=spike
> make newlib buid-sim
> ```
>
> Then, got following error:
>
> ``` [...]

I tried to reproduce this but get a different failure and, for some reason, the `pk` directory is empty (including no hidden items such as `.git`):
```
git clone https://github.com/riscv/riscv-gnu-toolchain multilib-spike
cd multilib-spike
./configure --prefix=`pwd`/installed-tools --enable-multilib --with-sim=spike

...

make[1]: Leaving directory '/home/tommy/Downloads/riscv-tools/multilib-spike/build-spike'
mkdir -p stamps/
date > stamps/build-spike
rm -rf stamps/build-pk32 build-pk32
mkdir build-pk32
cd build-pk32 && /home/tommy/Downloads/riscv-tools/multilib-spike/pk/configure \
	--prefix=/home/tommy/Downloads/riscv-tools/multilib-spike/installed-tools \
	--host=riscv64-unknown-elf \
	--with-arch=rv32gc
/bin/sh: 1: /home/tommy/Downloads/riscv-tools/multilib-spike/pk/configure: not found
make: *** [Makefile:825: stamps/build-pk32] Error 127

ls -al pk
total 8
drwxrwxr-x  2 tommy tommy 4096 Oct  5 13:47 .
drwxrwxr-x 30 tommy tommy 4096 Oct  5 17:22 ..
```
Branch: main
Commit: c2857f9c03867f9fe9ee8e44b7cf57dd83020455
Build log: [build.log.tar.gz](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9717376/build.log.tar.gz)

PR #1121 (2022-12-16): Quick fix sim pk build failure for RV32.
* Fix #1087.
* The riscv-pk repo can only be built by rv32i.
* Change configure to rv32i to make SIM=spike option happy for multilib.
closed 2022-12-16 by kito-cheng