riscv-collab/riscv-gnu-toolchain #1088: Wrong assembly for function calls in C

sumanth2002629, 2022-06-09:
[image]

This is the screenshot of the shell script that we have created to convert a C code to RISCV32IM ISA assembly code using RISCV-gnu-toolchain.

But, when we are calling functions in our C code, the assembly generated keeps on looping in the first function present in the C code without jumping to the called function.

Is there a specific way in which function calls should be made in the C code so that the assembly code generated is the correct one? 

The screenshots of the generated assembly code and the C program have been attached. The assembly code generated is looping between the addresses 0x28 and 0x2c instead of jumping to the called function.
Please look into this.

[image]
[image]

robertlipe, 2022-06-10:
You're showing a disassembly of the .o and not what's in the executable.  We can't see the fixups that are performed by the linker for that extern reference. Is &addition near in address space or is it far away in some separate address space in the linker?

What is this "cleaner" process and is it prepared to handle relocations in the absence of a call to the linker? (Since your call to objdump doesn't include -r it can't see the relocs anyway.) If cleaner is just looking at the disassembly of the object, it just doesn't have enough information to correctly build that call. 

If cleaner is being excessively clever (e.g. not calling ld and is parsing the .txt and building an executable from the opcode stream) try teaching it about relocs. 

P.S. reproducible test cases (in attachments or in the message itself) are usually more helpful than pictures of I/O.

TommyMurphyTM1234, 2022-09-27:
@sumanth2002629 - can you please follow up with additional info as requested if this is still a live issue? Thanks.

TommyMurphyTM1234, 2022-09-30:
Insufficient info provided to reproduce.

closed 2022-09-30 by TommyMurphyTM1234