riscv-collab/riscv-gnu-toolchain #1089: Unsupported rounding modes for floating point convert instructions.

pawks, 2022-06-15:
The specification mentions a `rm` field for all convert instructions where the width of the source operand is smaller than the target(like `fcvt.d.s`). However, the assembler does not allow specifying the `rm` for such instructions. I get the following error
```
Error: illegal operands `fcvt.d.s f11,f11,dyn'
``` 
The spec also mentions that the result of the instruction is not influenced by the rounding mode. However, the resulting binary encodings are influenced by it(currently the instructions have `7` as the default). Having support for specifying the rounding mode in the instruction will allow for testing all the features/behaviours specified in the ISA.

a4lg, 2022-07-01:
I support the idea in the disassembler but allowing them in the assembler can be a problem.

Since round-trip conversion is not strictly guaranteed in disassembler/assembler pair, how about this:
*   On disassembler, support non-`dyn` rounding modes on such instructions (*MAY* print the rounding mode, at least print as a *seemingly* valid instruction (considering following assembler requirement))
*   On assembler, make using non-`dyn` rounding modes on such instruction cause a warning or an error.  If only a warning is generated, corresponding rounding mode is applied.

pawks, 2022-07-01:
> I support the idea in the disassembler but allowing them in the assembler can be a problem (as @aswaterman pointed out).
> 
I don't think it has any negative consequences, since the spec says that the value of the field doesn't have any influence on the outcome. Software which generates assembly should ideally never be generating these instructions with explicit rounding modes. Having support for this in the assembler is a nice to have and I think gcc should have complete support for all the fields defined in the spec.

a4lg, 2022-07-02:
I personally agree to you @pawks but some might think this is too much.
So, I'm going from conservative approach (at least, we can agree that we definitely need a disassembler support) and discuss about assembler support later.

I guess that the proper place to discuss is riscv-non-isa/riscv-toolchain-conventions. Am I right? >@aswaterman, @kito-cheng

aswaterman, 2022-07-02:
Kito should comment about process. I’m just here to offer my opinion, not to represent any official position :)

TommyMurphyTM1234, 2023-10-18:
No update in over a year and this:

> I guess that one of the proper places to discuss is [riscv-non-isa/riscv-toolchain-conventions](https://github.com/riscv-non-isa/riscv-toolchain-conventions). Am I right? >@aswaterman, @kito-cheng

suggests that it's an issue that needs to be thrashed out elsewhere and not in the `riscv-gnu-toolchain` Issues.
Consequently, I am closing it.

closed 2023-10-18 by TommyMurphyTM1234