riscv-collab/riscv-gnu-toolchain #1095: How to add support for a new instruction?

yslys, 2022-06-25:
I would like to test a new instruction, for instance, with mnemonic ```newInst %a0, %a1```, and test it in gem5 hardware simulator. I have finished the decoder part of the gem5 hardware simulator, which makes the simulated hardware able to decode such new instruction. However, while generating the executable to be tested, I encountered this problem - I could not add such instruction into the program.

Typically what we will do is to use inline assembly like:
```
__asm__("newInst %0, %1": /* etc... */);
```
However, the cross compiler cannot recognize the new instruction. That is why I open this issue. Could anyone give me any hints on this? I have never worked with compiler code. May I know which file/directory is the best place to start?

yslys, 2022-06-25:
By the [answer to this question on StackOverflow](https://stackoverflow.com/questions/72751432/how-to-inline-assembly-code-using-bits-hard-code-in-c/72751456#72751456): I know the solution - inlining the raw bytes.

But I am still curious about how to modify the compiler to support a new instruction. Any hints would be appreciated.

JunningWu, 2022-09-28:
The easist way is Using `.insn`, which means you just need to write .word for your intruction.

[https://junningwu.haawking.com/tech/2019/11/28/%E4%BD%BF%E7%94%A8Gem5%E8%87%AA%E5%AE%9A%E4%B9%89RISC-V%E6%8C%87%E4%BB%A4%E9%9B%86-%E6%8C%81%E7%BB%AD%E6%9B%B4%E6%96%B0/](url)

TommyMurphyTM1234, 2022-09-28:
> By the [answer to this question on StackOverflow](https://stackoverflow.com/questions/72751432/how-to-inline-assembly-code-using-bits-hard-code-in-c/72751456#72751456): I know the solution - inlining the raw bytes.
> 
> But I am still curious about how to modify the compiler to support a new instruction. For instance, I think both the front end and back end should be modified. Any hints would be appreciated.

If you search the existing open and closed issues then you may find some useful info (most likely thanks to @jim-wilson) as this question has definitely been asked before.

TommyMurphyTM1234, 2022-11-10:
This is a list of all issues logged that I can find that relate to adding new instructions to the toolchain:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1095
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1055
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/837
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/782
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1003
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/795
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/660
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/543
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/341
* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/115

At this stage in the evolution of RISC-V,the majority of instruction addition will be done under the auspices of the relevant groups working on standardising, prototyping, and implementing/enhancing RISC-V extensions and their corresponding instructions. As such, there should be less of a demand for others to do so.

However, where somebody still wants to implement a non stadard custom extension/instruction here are some useful resources:

* GCC internals documentation: https://gcc.gnu.org/onlinedocs/gccint/
* https://hsandid.github.io/posts/risc-v-custom-instruction/
* https://ossg.bcs.org/blog/2020/10/29/adding-an-instruction-to-the-gnu-assembler/
* https://embdev.net/topic/228773
* https://groups.google.com/u/0/a/groups.riscv.org/g/sw-dev/c/sL_OHXYj3LY/m/NOm5Fwa3BAAJ
* https://forums.sifive.com/t/adding-custom-instruction-to-riscv-toolchain/4044/2

closed 2022-11-10 by TommyMurphyTM1234