riscv-collab/riscv-gnu-toolchain #1096: riscv-gnu-toolchain doesn't recognize the pmp configuration CSRs from 4 to 15

Abdulwadoodd, 2022-06-28:
In the latest privilege spec of risc-v, there are total 16 PMP configuration CSRs (pmpcfg0, pmpcfg1, pmpcfg2. ...., pmpcfg15). I'm able to compile only the first four of them and the toolchain does not recognize the rest of the CSRs. Can anyone comment if the support of all 16 PMP configuration CSRs is already added to the toolchain or not?

pz9115, 2022-07-06:
I tested with the binutils upstream, it's ok to use pmpcfgx[4,15] with Nelson's supports, the 2.38 release is a little early that unsupports those csrs. So I think you can checkout your binutils to the upstream to fix the problem.

Abdulwadoodd, 2022-09-16:
Thanks, the latest commit of binutils has all the privileged CSRs.

closed 2022-09-16 by Abdulwadoodd