riscv-collab/riscv-gnu-toolchain #1097: /lib64/lp64d/libc.so.6: version `GLIBC_2.33' not found

apepkuss, 2022-06-29:
Hi guys, I'm trying to build `riscv-gnu-toolchain` on the `qemu + fedora` environment. The steps I used are

```bash
./configure --prefix=/opt/riscv
make linux -j $(nproc)
```

I got the following error:

```bash
/opt/riscv/riscv64-unknown-linux-gnu/bin/ld: /home/riscv/workspace/riscv-gnu-toolchain/build/build-gcc-linux-stage2/./gcc/liblto_plugin.so: error loading plugin: /lib64/lp64d/libc.so.6: version `GLIBC_2.33' not found (required by /home/riscv/workspace/riscv-gnu-toolchain/build/build-gcc-linux-stage2/./gcc/liblto_plugin.so)
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:995: libgcc_s.so] Error 1
make[3]: Leaving directory '/home/riscv/workspace/riscv-gnu-toolchain/build/build-gcc-linux-stage2/riscv64-unknown-linux-gnu/libgcc'
make[2]: *** [Makefile:14660: all-target-libgcc] Error 2
make[2]: Leaving directory '/home/riscv/workspace/riscv-gnu-toolchain/build/build-gcc-linux-stage2'
make[1]: *** [Makefile:1038: all] Error 2
make[1]: Leaving directory '/home/riscv/workspace/riscv-gnu-toolchain/build/build-gcc-linux-stage2'
make: *** [Makefile:415: stamps/build-gcc-linux-stage2] Error 2
```

How should I fix the error? Thanks a lot!

apepkuss, 2022-07-04:
Hi @kito-cheng, 
I followed the steps described in the [Booting 64-bit Fedora](https://wiki.qemu.org/Documentation/Platforms/RISCV) section, but I'm not sure if the glibc's version is old or not. How should I check the version of glibc on my side? What's the minimum version of glibc? Thanks a lot!

TommyMurphyTM1234, 2022-08-15:
> How should I check the version of glibc on my side? 

`ldd --version`

> What's the minimum version of glibc?

For what exactly?

TommyMurphyTM1234, 2022-09-27:
Is this still a live issue @apepkuss ?

apepkuss, 2022-09-28:
> Is this still a live issue @apepkuss ?

Thanks for your reminder. I'll close the issue. Thanks!

closed 2022-09-28 by apepkuss