riscv-collab/riscv-gnu-toolchain #1099: RV32I soft float library attempts to access a floating point CSR (gcc-12 branch)

Wren6991, 2022-07-06:
I encountered this whilst using the gcc-12 branch to experiment with my RV32IMACZb* implementation. The symptom is that libgcc soft float attempts to access a floating point CSR, even though I didn't specify the F extension when I configured the toolchain.

### Repro

Build an RV32I ilp32 toolchain from the gcc-12 branch:

```bash
git clone --recursive --branch gcc-12 git@github.com:riscv-collab/riscv-gnu-toolchain.git
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv/riscv32-gcc12 --with-arch=rv32i --with-abi=ilp32
make -j$(nproc)
```

Create a file `f.c`:

```c
void main() {
	volatile float p, q, n;
	n = p / q;
}
```

Compile, link and disassemble:

```bash
/opt/riscv/riscv32-gcc12/bin/riscv32-unknown-elf-gcc -march=rv32i f.c -o f
/opt/riscv/riscv32-gcc12/bin/riscv32-unknown-elf-objdump -d f | less
```
### Expected behaviour

Code reachable from main will contain only RV32I instructions.

### Actual behaviour

The executable contains soft float routines as expected, but looking at the disassembly of `__divsf3`:

```
000101cc <__divsf3>:
   101cc:       fd010113                addi    sp,sp,-48
   101d0:       01612823                sw      s6,16(sp)
   101d4:       02112623                sw      ra,44(sp)
   101d8:       02812423                sw      s0,40(sp)
   101dc:       02912223                sw      s1,36(sp)
   101e0:       03212023                sw      s2,32(sp)
   101e4:       01312e23                sw      s3,28(sp)
   101e8:       01412c23                sw      s4,24(sp)
   101ec:       01512a23                sw      s5,20(sp)
   101f0:       01712623                sw      s7,12(sp)
   101f4:       00058b13                mv      s6,a1
   101f8:       002029f3                frrm    s3
...
```

The soft float code attempts to access the `frm` CSR, which is only present when the F extension is implemented.

### Troubleshooting

The implementation for `__divsf3` on RISC-V seems to be this one:

https://github.com/riscv-collab/riscv-gcc/blob/ca312387ab141060c20c388d83d6fc4b2099af1d/libgcc/soft-fp/divdf3.c#L43

I believe this is the correct expansion for that macro:

https://github.com/riscv-collab/riscv-gcc/blob/ca312387ab141060c20c388d83d6fc4b2099af1d/libgcc/config/riscv/sfp-machine.h#L113-L126

So this CSR access seems to be included because `__riscv_flen` is set when building RV32I libgcc, and indeed replacing that `#ifdef` with an `#if 0` makes my symptom go away.

`__riscv_flen` should only be set when `TARGET_HARD_FLOAT` is set (in `riscv-gcc/gcc/config/riscv/riscv-c.cc`), but I am struggling to find out where `TARGET_HARD_FLOAT` is actually getting set. It is surprising that `TARGET_HARD_FLOAT` would be set after configuring an RV32I ilp32 toolchain build.

I raised this issue here since it seems likely this is a configuration issue, not a problem with libgcc.

Wren6991, 2022-07-06:
> replacing that #ifdef with an #if 0 makes my symptom go away.

Or at least, it removes that `frrm`, but there are still floating point load/stores in the soft float code. It's almost as though the top-level configure is not propagating at all to the libgcc build.

TommyMurphyTM1234, 2022-08-07:
Maybe check the tools build log in case, in spite of `configure ... --with-arch=rv32i --with-abi=ilp32` the newlib libraries are getting built with floating point enabled for some reason?

Wren6991, 2022-09-06:
Thanks. GCC12 was promoted to master recently, and that branch does not show this problem, so I'm guessing this was just a configuration issue with that unstable GCC12 branch, and won't look into it any further. This can be closed.

closed 2022-09-06 by Wren6991