riscv-collab/riscv-gnu-toolchain #1100: Freebsd 13.1 compile error undefined reference to `srchilite::SourceHighlight::highlight(std::istream&, s

rcina, 2022-07-13:
When building from source on Freebsd 13.1 I am experiencing an error during compile of gdb regarding an undefined reference in SourceHighlight:

 CXX    init.o
  CXXLD  gdb
/usr/local/bin/ld: source-cache.o: in function `source_cache::ensure(symtab*)':
/hdd/builds/riscv-gnu-toolchain/riscv-gdb/gdb/source-cache.c:213: undefined reference to `srchilite::SourceHighlight::highlight(std::istream&, std::ostream&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/local/bin/ld: /hdd/builds/riscv-gnu-toolchain/riscv-gdb/gdb/source-cache.c:207: undefined reference to `srchilite::SourceHighlight::SourceHighlight(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
gmake[3]: *** [Makefile:1866: gdb] Error 1
gmake[3]: Leaving directory '/hdd/builds/riscv-gnu-toolchain/build-gdb-newlib/gdb'
gmake[2]: *** [Makefile:10065: all-gdb] Error 2
gmake[2]: Leaving directory '/hdd/builds/riscv-gnu-toolchain/build-gdb-newlib'
gmake[1]: *** [Makefile:853: all] Error 2
gmake[1]: Leaving directory '/hdd/builds/riscv-gnu-toolchain/build-gdb-newlib'
gmake: *** [Makefile:530: stamps/build-gdb-newlib] Error 2

I have the source-highlight package installed when this error occurs.  However, I am able to workaround the issue and get everything to compile correctly if I remove the source-highlight package.  Not sure why the definition of `srchilite::SourceHighlight::highlight is undefined and causing the build to fail.  I thought I would post the issue and give the work around I found for those users using Freebsd.

cmuellner, 2022-09-29:
We have just bumped GDB to version 12.1.
Is this still an issue?

rcina, 2022-10-04:
No it is no longer an issue. It seems to compile fine now. Thanks.

TommyMurphyTM1234, 2022-10-04:
Thanks for the update @rcina

closed 2022-10-04 by TommyMurphyTM1234