riscv-collab/riscv-gnu-toolchain #1101: How can user find out which extension supported when use riscv-gnu-toolchain without a source code sight?

pz9115, 2022-07-15:
When user use riscv-gnu-toolchain, a common problem is that they don't know what extensions was supported for their toolchain.
Should we add some feature for toolchain that make user use more convenient.

cmuellner, 2022-07-18:
For me the output of `gcc -v` is enough to find the corresponding sources.
However, I agree that this is not a user-friendly way.

What helps with ABIs and code models is `gcc --target-help`, but this does not list all acceptable march substrings.

BTW, AArch64 supports this:

```
echo | aarch64-linux-gnu-gcc -march=armv8.42-a -dM -E -
cc1: error: unknown value ‘armv8.42-a’ for ‘-march’
cc1: note: valid arguments are: armv8-a armv8.1-a armv8.2-a armv8.3-a armv8.4-a armv8.5-a armv8.6-a armv8-r; did you mean ‘armv8.2-a’?
```

And on x86 it is possible to get a list of supported march strings using `gcc -v --help`.

However, given the fact, that users can compose the march string on RISC-V, someone would have to write a patch, that collects all supported extensions and prints them (preferably as part of `--target-help`).

kito-cheng, 2022-08-04:
A potential proposal is adding `-mtune=?`, `-mcpu=?` and `-march=?` to list all supported option for those options, clang already supported first 2 option, so I think it would be great create some issue on https://github.com/riscv-non-isa/riscv-toolchain-conventions

kito-cheng, 2022-08-04:
Issue created https://github.com/riscv-non-isa/riscv-toolchain-conventions/issues/24

TommyMurphyTM1234, 2022-10-02:
Maybe I'm looking at this too simplistically, but doesn't the following provide the required info?
```
riscv64-unknown-elf-gcc -dumpspecs | grep arch
--traditional-format %(subtarget_asm_debugging_spec) %{fpie|fPIE|fpic|fPIC:-fpic} %{march=*} %{mabi=*} %{mno-relax} %{mbig-endian} %{mlittle-endian} %(subtarget_asm_spec)%{misa-spec=*}
-melf%{march=rv32*:32}%{march=rv64*:64}lriscv %{mno-relax:--no-relax} %{mbig-endian:-EB} %{mlittle-endian:-EL} %{shared}
march=rv64imafdv_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b mabi=lp64d

riscv64-unknown-elf-gcc -print-multi-lib
...
```
That provides the default arch/abi for the toolchain (last line of the grepped `-dumpspecs` ouput) and the set of other arch/abi pairs supported by the available multilibs.

TommyMurphyTM1234, 2023-10-15:
No update in over a year and it seems to me that earlier posts explain how to obtain the relevant info so I am closing this issue.

closed 2023-10-15 by TommyMurphyTM1234