riscv-collab/riscv-gnu-toolchain #1102: relocation truncated to fit: R_RISCV_PCREL_HI20 against `.LC19'

SpriteOvO, 2022-07-18:
I'm building `opencv` package on Arch Linux RISC-V 64-bit, with `gcc` version 12.1.0, and I get an error message:

```
/tmp/cc1A4RvS.ltrans3.ltrans.o: in function `.L1231':
<artificial>:(.text._ZN2cv8datasets9TR_svtImp8xmlParseERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERSt6vectorINS_3PtrINS0_6ObjectEEESaISD_EE.constprop.0+0x394): relocation truncated to fit: R_RISCV_PCREL_HI20 against `.LC19'
collect2: error: ld returned 1 exit status
```

The build script is https://github.com/archlinux/svntogit-packages/blob/8d833c937701952d9c97ff329d9b987ba6e5b10f/trunk/PKGBUILD and needs apply a patch https://github.com/felixonmars/archriscv-packages/blob/d99ef9f335c5647666f04a52a8cb766cfdf0b18e/opencv/riscv64.patch.

If I add a line `options=(!lto)` to that script, which means passing argument to gcc to disable LTO, the build will complete successfully.

Failed build log: [opencv-4.6.0-3-riscv64-build.log](https://github.com/riscv-collab/riscv-gnu-toolchain/files/9135778/opencv-4.6.0-3-riscv64-build.log)

Sorry for my lack of experience with `gcc` debugging, I can't make a minimal reproducible example. Let me know if you need more information and I'll do my best to provide it.

TommyMurphyTM1234, 2022-10-02:
This seems to be a duplicate of https://github.com/riscv-collab/riscv-gnu-toolchain/issues/624 ?

TommyMurphyTM1234, 2022-12-15:
Duplicate of https://github.com/riscv-collab/riscv-gnu-toolchain/issues/624

closed 2022-12-15 by TommyMurphyTM1234