riscv-collab/riscv-gnu-toolchain #1103: Different assembly for system calls.

sumanth2002629, 2022-07-20:
We are using riscv-gcc toolchain to convert C program to risc-V assembly. But the generated assembly has 16bit instructions. Not sure why? Please help.

These are the commands used to generate assembly:
riscv32-unknown-elf-gcc -mabi=ilp32 -march=rv32i -o sys_call sys_call.c 
riscv32-unknown-elf-objdump -d -r sys_call

And few lines of assembly code in the screenshot.
[image]

stnolting, 2022-07-20:
Maybe your library code (for things like `exit`) has been compiled with `rv32ic`?
What does the configuration of your toolchain look like (e.g. `riscv32-unknown-elf-gcc -v`)?

sumanth2002629, 2022-07-20:
Yes it is configured  `--with-arch=rv32imc`

stnolting, 2022-07-22:
> Yes it is configured `--with-arch=rv32imc`.

I am no expert here, but I think that is the cause why your system libs were compiled using the `C` extension.

> So should I change the configuration to `rv32im` only??

This should work.

Btw, you might want to use the `multilib` option when configuring/compiling the toolchain - I think this will give you all kind of ISA ext. combinations.

TommyMurphyTM1234, 2022-08-11:
> Btw, you might want to use the `multilib` option when configuring/compiling the toolchain - I think this will give you all kind of ISA ext. combinations.

Several, but definitely not all... :-)

https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/riscv/t-elf-multilib

TommyMurphyTM1234, 2022-09-27:
Questions about how to build the toolchain for a particular target arch - not an issue with the RISC-V GCC toolchain itself.